microsoft/vscode-arduino #867: Serial monitor not wok on FTDI

CodeNameHawk, 2019-07-15:
I use an USB-COM adapter FTDI FT232RL with control DTR and RTS, to automatic programming ESP8266
Unfortunately, the plugin incorrectly sets the levels to DTR and RTS, which does not allow to use normally Serial Monitor

CodeNameHawk, 2019-07-15:
Version: 1.36.1 (user setup)
Commit: 2213894ea0415ee8c85c5eea0d0ff81ecc191529
Date: 2019-07-08T22:58:59.274Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT ia32 10.0.18362

Arduino 0.2.27

zunhu, 2019-08-20:
can you try to not use automatic programming ESP8266 and select right programming?
sometimes auto selected was wrong.

CodeNameHawk, 2019-08-20:
Automatic programming normal work , but after programning not work Serial Monitor.

gcampbell-msft, 2023-02-16:
@CodeNameHawk We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue!

closed 2023-02-17 by CodeNameHawk