microsoft/vscode-arduino #869: Exit with code=undefined

thangiot, 2019-07-21:
I've got the same problem #805 but I cannot solve my problems with these instructions in there. I used Arduino 1.8.9, vscode 1.36.1, Windows 10, Arduino Path: "C:\Program Files (x86)\Arduino" . Please help me, thanks.
```
[Starting] Verify sketch - Blink.ino
Please see the build logs in Output path: c:\Users\tqtha\OneDrive\Documents\Workspace\Arduino_Projects\build
[Error] Exit with code=undefined
```

arduino.json
```
{
    "sketch": "Blink.ino",
    "port": "COM3",
    "board": "arduino:avr:nano",
    "output": "./build"
}
```

c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}\\**",
                "C:/Program Files (x86)\\Arduino\\hardware\\tools\\avr\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\**",
                "C:\\Program Files (x86)\\Arduino\\tools\\**"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 4
}
```
settings.json
```
{
    "arduino.path": "C:\\Program Files (x86)\\Arduino", 
    "arduino.additionalUrls": "", 
    "arduino.logLevel": "info",
    "arduino.commandPath": "arduino_debug.exe"
}
```

_Originally posted by @thangiot in https://github.com/microsoft/vscode-arduino/issues/805#issuecomment-513524686_

KThen, 2019-08-01:
I've got the same issue. Someone please look into this.

zunhu, 2019-08-01:
@thangiot @KThen. have you try these steps? in #805

1.Open Arduino IDE to update board packages and compile the project.
2.Open Arduino Board Manager in VS Code Arduino extension to see if the page load correctly.
3.Arduino CLI may be blocked by Windows firewall. A file called arduino_debug.exe should be in your Arduino IDE installation folder, and VS Code extension calls this file to execute Arduino commands. This file needs to access to the Internet.

KThen, 2019-08-01:
1. I am able to compile from within Arduino IDE.
2. I am able to open Arduino Board Manager in VS Code Arduino extension.
3. I have already whitelisted both arduino.exe and arduino_debug.exe in Windows Defender Firewall.

I've used many times this extension in the past and have no issues with it, this just occurs recently. I can't recall exactly when it happened, but I suspect it is not a problem with this extension, it might have something to do with recent VS Code updates.

KThen, 2019-08-01:
[image]

Does it help if I attach a debug message? I don't think it is relevant but maybe it will give clues?

Here on log.ts line 173 is the original error. I added a line to print out arduinoContext_1.default in serialMonitor.js and I noticed _arduinoApp, _debuggerManager and _boardManager are all null.

PeterWone, 2019-09-22:
I am experiencing _exactly_ as @KThen describes it, right down to the null reference in SerialMonitor.initialize 

I have ripped out VSCode and including all the files it leaves behind, and reinstalled it and the Arduino extension. I installed the Arduino IDE in version 1.8.8 because that version is blessed in the documentation, and was able to compile and deploy using that.

I have the same set up on my notebook and it works perfectly there. The notebook doesn't have build 1903 because Microsoft is blocking it pending resolution of driver issues for Surface Book 2, but my desktop has been on 1903 for months now and this problem is quite recent.

PeterWone, 2019-09-23:
It works perfectly in a freshly built VM with Win10 1903 and latest everything. A key observation is that on my problem machine there is no output from log4j whereas on the fresh VM, output looks like this
```
[Starting] Verify sketch - Debounce.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Verify could be slow. See README.
Set log4j store directory C:\Users\peter\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
```
This implies that there are settings in force on my problem machine suppressing logs among other things, which survived reinstallation of the Arduino IDE. In fact on that workstation I never saw log4j messages even when it _was_ working. Any thoughts on what controls this?

PeterWone, 2019-09-23:
I spun up the extension in debug mode and `util.ts:211` is barfing when it tries to resolve the codepage for Win32

```            codepage = childProcess.execSync("chcp").toString().split(":").pop().trim();```

I broke up the parts of this expression and determined that the problem occurs in `execSync("chcp")` and then found that the error message is useless because reason doesn't have a code property. Inspection shows that it _does_ have a message property so I changed the calling context's error trap 
```
 } catch (reason) {
    arduinoChannel.error(reason.code ? `Exit with code=${reason.code}${os.EOL}` : reason.message);
    return false;
 }
```
This returns a useful message 
```
[Error] Command failed: chcp
'chcp' is not recognized as an internal or external command,
operable program or batch file.
```

WTF is `chcp`? The interwebs says it's an ancient utility for setting the code page, which when invoked without a parameter can be used to determine the current code page. Switching to a command window I type in chcp and get the above error. Switching to a box on which the original problem does not occur, I find that chcp works. More investigation reveals my path has been annihilated. The only thing on it is NodeJS, and the only reason _that_ is present is I reinstalled it because it wasn't working. 

How my path got broken is out of scope for this discussion. The key insight is that repairing it caused chcp to work resolving the code=undefined thing. The error handling is broken but the real problem is that unavailability of chcp makes the whole process fail. A reasonable fallback would be to default to 850 with a warning.

I will do a PR.

dooriya, 2019-09-26:
@PeterWone Thank you very much for your active participation and contribution. 

Before getting approval for your [Pull Request](https://github.com/microsoft/vscode-arduino/pull/899), could you please fix the build failure with the Travis CI?

[image]

Please try  fixing the `.travis.yml` file as below:
[image]

PeterWone, 2019-09-26:
Please look at #900 because the problem is your Linux build environment, not my code. Maybe you haven’t noticed but the main branch is failing in the same way for the same reason.

When I hand built a test environment using the latest Ubuntu I got the failure you want me to correct.
When I hand built a test environment using the latest STABLE Ubuntu all tests passed.

The problem is that the Arduino IDE does not work on the latest Ubuntu. It works fine on the LTS build.

I looked at your build config and it shouldn’t be using the latest Ubuntu, but there’s something wrong. Maybe you should take a look at the Docker image.

[image]<https://user-images.githubusercontent.com/10163840/65672412-96f8d580-e07b-11e9-9a8c-001a33ed4f2a.png>

[image]<https://user-images.githubusercontent.com/10163840/65672515-cad3fb00-e07b-11e9-82e1-c327ff49d814.png>

dooriya, 2019-09-26:
We've just fixed the Travis CI in master branch.
@PeterWone  Could you please merge the latest changes from master into your PR?

PeterWone, 2019-09-26:
Wow, that was quick! I'm on it.

PeterWone, 2019-09-26:
@dooriya This is the first time I've had to sync changes like this, I hope I've done this right.

dooriya, 2019-09-26:
@PerterWone You make it :) I approved your PR, please feel free to merge it into master. We’ll add it into our next release.

Thanks again for the contributions.

PeterWone, 2019-09-26:
@dooriya -- you say "feel free to merge it into master". Are you telling me that I have the necessary permissions to merge into microsoft/vscode-arduino ?

What step should I now take? I do not know how to proceed.

dooriya, 2019-09-26:
@PeterWone  Sorry for that, it should be us to merge the PR.

PeterWone, 2019-09-26:
Not at all -- my ignorance of the workings of GitHub is hardly your fault. Thank you for helping me to get this done.

dooriya, 2019-09-26:
Will update you once the new release is delivered.

dooriya, 2019-11-21:
PR is merged with the fix, so close it here.
Please feel free to re-open this issue if you still get the problem.

dooriya, 2019-11-21:
The fix is released in Arduino 0.2.28.

closed 2019-11-21 by dooriya