microsoft/vscode-arduino #870: cannot open source file intellisense error for file that is never openned

mMerlin, 2019-07-22:
It would be nice if the intellisense parse did not complain about not being able to open include files that the compiler never accesses anyway.  There are some issues posted that sound similar, but were due to problems with the `includePath` entry in `c_cpp_properities.json`.  This is not.  The referenced file does not exist at all, and does not need to exist for the current environment.

```txt
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/home/phil/development/wcrs/repos/nRF2401-Joy2/nRF2401-Joy2.ino).
cannot open source file "utility/includes.h" (dependency of "RF24.h")
```

However, in `RF24.h` is…

```cpp
#if defined (RF24_LINUX) || defined (LITTLEWIRE)
  #include "utility/includes.h"
#elif defined SOFTSPI
  #include <DigitalIO.h>
#endif
```

and the current environment does not define either `RF24_LINUX` or `LITTLEWIRE`.

The parser is obviously tracing the nested includes.  Can't it track the defines as well?  Probably not near as easy as it sounds.

A common case for this, is the WProgram.h include file that was used prior to the Arduino IDE version 1.0.  There are (or at least used to be) libraries that have if/else defs to include either that or the replacement Arduino.h, depending on the environment version.  Which is ALWAYS going to generate bogus `cannot open` messages.  Only one of the files in the #if #else are going to exist in a single environment.

closed 2019-08-26 by stale[bot]

haizaar, 2019-09-27:
I just landed here searching for solution about allegedly missing "WProgram.h". Any chance this can be fixed?

hellyzh, 2019-10-31:
Tested from my side, the parser is tracking the defines, and not give include error. Please check installed the latest [C/C++ IntelliSense tool](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools)
[image]

nmaupu, 2019-12-12:
I have the exact same problem even after toggling around in C/C++ intellisense options...