microsoft/vscode-arduino #871: Not finding header files  - is it reading c_cpp_properties?

ImolaS3, 2019-07-27:
Hopefully someone can help - I have spent 2 weeks trying to fix this and exhausted pages of google results :(

I use a Mac mostly these days and wanted a nice "proper" IDE for Mac and VSCODE seemed to fit the bill.

I have it working fine if I place my custom headers in the actual project folder and use"" to point to them, but when placed in another folder using <>, it always fails.

I have added the path to the c_cpp_properties.json and played with "Default" and "Tag Parser" intellisense.

The Colours header in this example is located in "/Users/antonysutton/Arduino_H/" and called Colours.h (I have also tried ""/Users/antonysutton/Arduino_H/**"

When I try to compile, the error is "TonyTouchScreen:7:66: error: Colours.h: No such file or directory"

Headers in code:
#include <Adafruit_GFX.h>
#include <MCUFRIEND_kbv.h>
#include "TouchScreen_kbv.h"
#include <Colours.h>

Using the following syntax compiles fine, so it si not looking in the includepath unless explicitly stated :

#include <Adafruit_GFX.h>
#include <MCUFRIEND_kbv.h>
#include "TouchScreen_kbv.h"
#include </Users/antonysutton/Arduino_H/Colours.h>

c_cpp_properties.json

{
"configurations": [
{
"name": "Mac",
"includePath": [
"/Applications/Arduino.app/Contents/Java/tools/",
"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/",
"/Users/antonysutton/Arduino_H/",
"${workspaceFolder}/"

        ],
        "defines": [],
        "intelliSenseMode": "clang-x64",
        "browse":{
            "path":[
                "/Applications/Arduino.app/Contents/Java/tools/**",
                "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/**",
                "/Users/antonysutton/Arduino_H/**",
                "${workspaceFolder}/**"
            ]
        },
        "macFrameworkPath": [
            "/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks"
        ],
        "compilerPath": "/usr/bin/clang",
        "cStandard": "c11",
        "cppStandard": "c++17",
       
        "forcedInclude": [
            "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h"
        ]
    }
],
"version": 4
}

Settings.json
"C_Cpp.intelliSenseEngine": "Tag Parser",

ImolaS3, 2019-07-27:
I am pretty sure the3 compilation is not even looking at c_cpp_properties.json as whatever i do to it, the error remains the same, none added, none taken away even if i delete the .json file it still has exactly the same error

Someone please help :(

brae-pete, 2019-08-24:
So I did two things to get rid of that error. I ran the Arduino library manager (Ctl + Shift + P >> Arduino: Library Manger). Then I ran the Arduino Board configuration and selected my arduino (Ctl + Shift + P). After I did that it was able to find the default libraries. I am not actually compiling and uploading to my arduino using VS-code, I just use it to look through files. But you should be able to verify and upload using the "Verify" and "Upload" commands. 

Commands listed at  [https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino]
This extension provides several commands in the Command Palette (F1 or Ctrl + Shift + P) for working with *.ino files:

    Arduino: Board Manager: Manage packages for boards. You can add 3rd party Arduino board by configuring Additional Board Manager URLs in the board manager.
    Arduino: Change Baud Rate: Change the baud rate of the selected serial port.
    Arduino: Change Board Type: Change board type or platform.
    Arduino: Close Serial Monitor: Stop the serial monitor and release the serial port.
    Arduino: Examples: Show list of examples.
    Arduino: Initialize: Scaffold a VS Code project with an Arduino sketch.
    Arduino: Library Manager: Explore and manage libraries.
    Arduino: Open Serial Monitor: Open the serial monitor in the integrated output window.
    Arduino: Select Serial Port: Change the current serial port.
    Arduino: Send Text to Serial Port: Send a line of text via the current serial port.
    Arduino: Upload: Build sketch and upload to Arduino board.
    Arduino: Upload Using Programmer: Upload using an external programmer.
    Arduino: Verify: Build sketch.
"

zunhu, 2019-08-29:
Hi, have you tried put Colours.h to same path with 
#include <Adafruit_GFX.h>
#include <MCUFRIEND_kbv.h> ?

zunhu, 2019-09-17:
dev is working on intellisense issue.

Jeff-Birt, 2019-09-23:
This bug makes the whole extension next to useless. Please advise as to a work around...

Jeff-Birt, 2019-10-07:
My VS Code installation started closing randomly so I removed it and downloaded the system installer version for windows (not local user installer) . Now the Arduino extension works properly, i.e. it finds the header files and will compile. Perhaps this is an issue with the local installer version??

dean-rowntree, 2019-10-11:
> My VS Code installation started closing randomly so I removed it and downloaded the system installer version for windows (not local user installer) . Now the Arduino extension works properly, i.e. [...]

I tried this and still had the same issue.
What I found to work was to have the following set (search for "tag parser" in settings):
[image]

Jeff-Birt, 2019-10-11:
Yes, the intellisense does not work. I gave up and moved to Visual Micro. It works...

fclante, 2019-10-27:
Hi none of the suggestions worked. 
      * When including custom libraries --> The plugin seemed to be able to F12 navigate to source files, 
      * When verifying the code it could not find the file.
	  
For some reason it only works when I put the custom .h and .cpp files
in C:\Users\[USERNAME]\Documents\Arduino\libraries folder.

hellyzh, 2019-10-31:
The C/C++ extension with c_cpp_properties.json is only used for syntax highlighting, smart suggestions, but not involved in compile, verify or build. It is expected as **fclante** mentioned above. The correct c_cpp_properties.json can make you F12 navigate to source files, but to verify/compile, you still need to ensure your custom .h and .cpp files in the project path or the system path that compiler already known.

terrillmoore, 2019-12-06:
It works for me, but I find I must be very careful with case in the paths in `c_cpp_properties.json`.

Found empirically on Windows 10:

1) the drive letter **must** be upper case.
2) the case of each element of the path must match exactly what is used is on disk.

For example, for me: 

- `c:/Users/MYNAMEHERE/AppData/Local/Arduino15/packages/mcci/hardware/stm32/**` doesn't work. 
- `C:/Users/MYNAMEHERE/AppData/Local/Arduino15/packages/mcci/hardware/stm32/**` does.

Similarly:

- `C:/Users/MYNAMEHERE/Documents/arduino/libraries/**` doesn't work
- `C:/Users/MYNAMEHERE/Documents/Arduino/libraries/**` does (upper case `A` in `Arduino`, as it is on my system).

Intellisense is also slow to pick up changes; I find that saving all files and then selecting Arduino as the C/C++ Configuration causes a rescan.

With these settings, and making sure the compiler path, `cStandard`, `cppStandard`, and `intelliSenseMode` settings are right, I get proper hinting and pretty good preview of compile errors. With the Arduino environment being so slow when compiling, that saves a lot of time.

hellyzh, 2019-12-06:
Yes, make sure all set correctly especially `compilerPath`, `intelliSenseMode` will work well to give smart suggestion.

hellyzh, 2020-01-06:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-01-06 by hellyzh

terrillmoore, 2020-01-06:
@hellyzh, is it not a bug that things are case sensitive on Windows? On the command line or in Exploror, the file system paths are not case sensitive. The drive letters are not case sensitive.  But clearly, in my testing, Intellisense setup is case sensitive. I note that @ImolaS3 was also using a mac, which like Windows is also case insensitive (for the default the file system setup).

hellyzh, 2020-01-07:
@terrillmoore, I have tested from my side, it is not case sensitive for windows file path. When you do the test, you can also try to clear the Intellisense cache by running `"C/C++: Reset intelliSense Database"` or restart VSCode. If you find it is really a bug, please feel free to open a new bug, thank you.

jrobertson-insite, 2020-06-23:
I found this setting to be key for me. In settings.json set "C_Cpp.intelliSenseEngine" to "Tag Parser"

`{
  "C_Cpp.intelliSenseEngine": "Tag Parser"
}`

MohammedNoureldin, 2020-10-30:
Have anyone found a solution for this? this issue is really a pain and makes the whole VS code for Arduino useless...

Milkiananas, 2023-05-09:
Try C/C++: Restart IntelliSense for Active File from the command palette.