microsoft/vscode-arduino #875: 中文乱码

Aaronwang81, 2019-07-29:
vscode + arduino 1.8.9
win7 64
输出窗口显示乱码
[image]

zunhu, 2019-08-01:
Hi @Aaronwang81  ,  try to modify with running system.

refer : https://forum.arduino.cc/index.php?topic=76658.0

Aaronwang81, 2019-08-01:
> Hi @Aaronwang81 , try to modify with running system.
> 
> refer : https://forum.arduino.cc/index.php?topic=76658.0

Sorry，It doesn't work。

Aaronwang81, 2019-08-01:
Maybe I known what happen。
The arduino_debug.exe output UTF-8 strings。（I redirect it to a txt file。）
But the VSCode output trade it as GBK。。。
How can I change it to UTF-8？

Aaronwang81, 2019-08-01:
In vscode-arduino/src/common/util.ts lines 209
Befault cp  is 65001
But，if os is win32，It will get the chcp（for me, it is 936）， not match arduino_debug.exe's output。
I think it's a bug。-_-

zfm076, 2019-08-09:
@Aaronwang81 问题解决了么？我也是编译过程信息乱码，设置了一遍还是不行。

cubex108, 2019-08-13:
There is a method here for reference. @Aaronwang81 
https://blog.csdn.net/weixin_45033342/article/details/90574434

he0119, 2019-08-14:
> There is a method here for reference. @Aaronwang81
> https://blog.csdn.net/weixin_45033342/article/details/90574434

This method fixed the problem. But it also prevents Final Fantasy XIV(China Version) from starting.😂

Guoliang-Zhong, 2019-08-17:
In this mothod, https://blog.csdn.net/weixin_45033342/article/details/90574434 ,changing to UTF-8 will cause many problems! like when open txt, it will shows error codes also.
so I setup language of Arduino IDE to English.

那个使用Bata版 utf-8的方法，会造成更多的问题！很多.txt和软件都会显示乱码！所以我还是换成了英文。

Aaronwang81, 2019-08-19:
找到vscode-arduino插件的安装目录，把util.js第215行注释掉就可以了。
前提是你用的arduino 1.8.9版本。
文件具体位置：x:\userdata\'yourusername'\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.27\out\src\common\
我是win7系统。如果路径不对，你们直接用文件工具找对应的文件名就可以了。改完重启vscode马上OK。

zunhu, 2019-08-26:
used workaround to mitigate this issue. issue closed.

closed 2019-08-26 by zunhu

BigMangos, 2020-03-31:
@Aaronwang81 请问 215 行是哪一行？

Aaronwang81, 2020-03-31:
> @Aaronwang81 请问 215 行是哪一行？

最新的0.2.29版本应该是218行，
codepage = chcp.toString().split(":").pop().trim()
但这是个公共函数，修改后可能引起其他输出显示 问题。如果有问题，请回退。这也是官方一直没修的原因之一吧。

BigMangos, 2020-04-01:
> > @Aaronwang81 请问 215 行是哪一行？
> 
> 最新的0.2.29版本应该是218行，
> codepage = chcp.toString().split(":").pop().trim()
> 但这是个公共函数，修改后可能引起其他输出显示 问题。如果有问题，请回退。这也是官方一直没修的原因之一吧。

不太懂 TS ，我把这些都注释了
```
if (os.platform() === "win32") {
    try {
    const chcp = childProcess.execSync("chcp.com");
    codepage = chcp.toString().split(":").pop().trim();
}
    catch (error) {
        outputChannel_1.arduinoChannel.warning(`Defaulting to code page 850 because chcp.com failed.\
        \rEnsure your path includes %SystemRoot%\\system32\r${error.message}`);
        codepage = "850";
    }
}
```
输出中文正常了，但是有些字显示方框，比如`地址`，非常奇怪，大部分字还是正常的

mzhboy, 2020-11-30:
%homepath%\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\src\common\util.js

注释掉部分代码后的部分日志，上传是成功的

> DEBUG StatusLogger LoggerContext[name=1e6f5c3, org.apache.logging.log4j.core.LoggerContext@2055d6] started OK.
> 正在准备开发板...
> 正在验证...
> 项目使用了 38416 字节，占用了 (58%) 程序存储空间。最大为 65536 字节。
> 全局变量使用了4600字节，(22%)的动态内存，余留15880字节局部变量。最大为20480字节。
> 上传...
> maple_loader v0.1
> Resetting to bootloader via DTR pulse
> Searching for DFU device [1EAF:0003]...
> Found it!
> 
> Opening USB Device 0x1eaf:0x0003...
> Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
> Setting Configuration 1...
> Claiming USB DFU Interface...
> Setting Alternate Setting ...
> Determining device status: state = dfuIDLE, status = 0
> dfuIDLE, continuing
> Transfer Size = 0x0400
> bytes_per_hash=768
> Starting download: [##################################################] finished!
> error resetting after download: usb_reset: could not reset device, win error: ϵͳ�Ҳ���ָ�����ļ���
> 
> state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
> 
> Done!
> Resetting USB to switch back to runtime mode

GF-Huang, 2021-02-21:
This issue should re-open until this plugin fixed.

starofys, 2021-03-09:
> This issue should re-open until this plugin fixed.

open **arduino_debug.l4j.ini** and modify
`-Dfile.encoding=GBK`

GF-Huang, 2021-03-09:
> > This issue should re-open until this plugin fixed.
> 
> open **arduino_debug.l4j.ini** and modify
> `-Dfile.encoding=GBK`

Where is the file.