microsoft/vscode-arduino #876: Missing #define ARDUINO 10808 logic?

supaplextor, 2019-07-29:
Same issue as reported here: https://github.com/microsoft/vscode-arduino/issues/829#issuecomment-502337491

Without the manual define, the includes don't work.

```
#define ARDUINO 10808
#include <Arduino.h>
#include <DHTesp.h>

DHTesp dht;

void setup() {
  Serial.begin(115200);
  dht.setup(D5, DHTesp::DHT11);
}

float temp = 0;
float ltemp = 1;

void loop() {
  delay(25);
  temp = dht.getTemperature();

  if ((!isnan(temp)) && (ltemp != temp)) {
    Serial.printf("%0.1f F\n", dht.toFahrenheit(temp));
    Serial.printf("%0.1f humidity\n", dht.getHumidity());
    if (temp != ltemp) {
      ltemp = temp;
    }
  }
}

```

```
supaplex@supaplex-N56JR:~$ dpkg -l code-insiders 
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  code-insiders  1.37.0-15641 amd64        Code editing. Redefined.

```

Ubuntu 18.04.2 LTS

supaplextor, 2019-08-16:
supaplex@supaplex-N56JR:~$ dpkg -l code-insiders
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  code-insiders  1.37.0-15650 amd64        Code editing. Redefined.

issue still present in this release.

supaplextor, 2019-09-01:
```

supaplex@supaplex-N56JR:~$ dpkg -l code-insiders
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                          Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  code-insiders                                 1.38.0-1566539276           amd64                       Code editing. Redefined.

```
also present

mikecentola, 2020-02-17:
This is still present in the latest release.  I too had to add the define.

elektronikworkshop, 2020-02-18:
This is due to the improper IntelliSense implementation of vscode-arduino. I'm currently working on this and a beta should be available soon at least for OSX and Linux. Windows will follow.

Anyone interested can track the progress [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md) and I often report progress to issue #438 as well.

Regards
EW

hellyzh, 2020-02-20:
@mikecentola , have you tried to set "#define ARDUINO 1080 logic" to file "c_cpp_properties.json"?

[image]

elektronikworkshop, 2020-02-20:
Hi @mikecentola 

if you are on OSX or Linux, you can try my alpha releases which are available since yesterday from [here](https://www.dropbox.com/sh/whmcdt26chyjgby/AAB1Ld2fzZ9Z_NfM3CRay17wa). I'll work on Windows support but this might take some days since I have to convince myself to launch my virtual maching :wink:

hellyzh, 2020-03-13:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-03-13 by hellyzh