microsoft/vscode-arduino #877: Status bar always active

jscheel, 2019-08-04:
Not sure if something has changed on my end, but my status bar always shows all of the arduino actions now, even in projects/files that have nothing to do with arduino.

zunhu, 2019-08-08:
@jscheel what's your OS, Arduino, Arduino extension version?

and can you please paste screenshot for it?

zunhu, 2019-08-20:
not sure if it is temporary issue, since long time no active, issue closed, please assign another issue if still happen.

closed 2019-08-20 by zunhu

jscheel, 2019-08-20:
MacOS 10.12.5

Name: Arduino
Id: vsciot-vscode.vscode-arduino
Description: Arduino for Visual Studio Code
Version: 0.2.27
Publisher: Microsoft
VS Marketplace Link: https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino

Interestingly, it only seems to appear in some windows. Here it is in a completely unrelated project on a .jsx (react) file.

[image]

jscheel, 2019-08-20:
@zunhu sorry for the delay, been in a hole ^^