microsoft/vscode-arduino #879: Weird arduino serial output

realivanjx, 2019-08-07:
When i compile any code that uses `Serial.print` it prints very weird message:

[image]

[image]

But when i copied all my code to the original arduino ide and upload it from there, it works just fine:

[image]

[image]

Why this happens?

zunhu, 2019-08-08:
@realivanjx what's your OS? 
looks similar with https://forum.arduino.cc/index.php?topic=76658.0

realivanjx, 2019-08-08:
Windows 10 latest

zunhu, 2019-08-20:
same issue with #875 , mark as bug.

zunhu, 2019-08-26:
1, try to find file C:\Users\username\.vscode\extensions\vsciot-vscode.vscode-arduino-0.2.27\out\src\common\util.js
2. commented out line:215.
3. restart vscode.

closed 2019-09-09 by zunhu