microsoft/vscode-arduino #880: Cannot set output folder using Environment Variable

dasfdlinux, 2019-08-08:
When attempting to set an Output folder in Windows using an environment variable, it is getting treated as a relative path. Am I using the environment variables incorrectly? I referenced [https://code.visualstudio.com/docs/editor/variables-reference#_environment-variables](https://code.visualstudio.com/docs/editor/variables-reference#_environment-variables) and #450 

arduino.json
`{
    "port": "COM5",
    "board": "arduino:avr:pro",
    "sketch": "wireless_temperature.ino",
    "programmer": "AVRISP mkII",
    "configuration": "cpu=8MHzatmega328",
    "output": "${env:TEMP}/arduinoBuild"
}`

Error:
> Please check the "output" in the sketch Settings.Cannot find the output path:c:\Users\dshaffer\Documents\Misc Code\wireless_temperature\${env:TEMP}\arduinoBuild

VS Code 1.36.1
Arduino 0.2.27
Windows 10

dilin-MS, 2019-08-20:
Hi, the `${variableName}` syntax is supported in VS Code's built-in `launch.json` and `tasks.json` files( as the [link](https://code.visualstudio.com/docs/editor/variables-reference#_environment-variables) suggests). Currently arduino extension's configuration file `arduino.json` does not support the environment variable.

dasfdlinux, 2019-08-20:
Thanks, I see that now. I clearly skimmed over the first paragraph on that page. This would be a great enhancement.

BeautifulNow97, 2019-09-11:
Cannot find the sketch file.
编译出现

closed 2019-10-16 by stale[bot]

tsbertalan, 2023-01-31:
Bumping this -- I want to be able to have one `arduino.json` file that both Windows and Linux machines use for compilations (or even Windows machines with different usernames), and have my output dir in like `~/tmp/arduino_output/`.