microsoft/vscode-arduino #882: error navigation

maziac, 2019-08-24:
This is a featured request.
Maybe this feature already exists then it does not work for me.

If an Arduino Sketch is verified and an error is found in the code it is not possible to easily navigate to the position in the Arduino code. I.e. it does not appear in the ‚Problems‘ section.

zunhu, 2019-08-26:
Hi Maziac, Arduino has highlights for error code. see screenshot. are you talking about this?
use scrollbar can go to the position  of code.

[image]

maziac, 2019-08-26:
How does it work?
Here is the screenshot from my project:
There are no squiggles.
[image]

Originally I was searching for a way similar to e.g. typescript (or C++) error/problem handling.
See following screenshots:
[image]

[image]

After "compiling" (transpiling) the 'problemMatcher' parses the compiler-output and creates 'problems' in the corresponding tab.
If selecting one of the problems the cursor is directly moved to that file and line.

zunhu, 2019-08-27:
Even you don't compiling\verfiy code, input squiggles that parser will show red wavy line.
If you need to select one of the problems and the cursor is directly moved to that file and line, you need to do compiling\verfiy code first, this is by design.

maziac, 2019-08-27:
Two issues here:
a) As you can see from my first screenshot: there are no squiggles from the parser.
b) I would like to see the error in the 'problems tab'.

For b): Is this somehow supported by the extension.
And for a): It does not work.

zunhu, 2019-08-29:
I see, I met same issue before (it is temporary issue), it works sometimes.
I marked as a bug, but not blocking issue.
for workaround, as I said you can do verify or compile code first.

maziac, 2019-08-29:
Could you maybe explain to what of the both issues you are answering. a) or b).
For me it seems that a) is not supported by the extension. Is this correct?
And for b): it does not change anything if I do a verify. Could you please give a step-by-step description.

zunhu, 2019-08-30:
a) squiggles will show and without any execution, I'm not sure it is Arduino Extension issue for VSCode issue. just one time no show during my testing.
b) Yes, it is in problem's tab, I misunderstanding before. no need to do verfiy.

[image]

closed 2019-09-09 by zunhu