microsoft/vscode-arduino #886: Integrated functions undefined

DoganM95, 2019-08-29:
### Hi, i am fighting with arduino for vscode for days now. it has a huge bug regarding header files:
### Sample of what i am trying:

[image]

### as you can see, the integrated functions of arduino ide are not working, like Serial.

### But as soon as i include any library from Library Manager (even the ones which have noting to do with Serial), the internal functions get recognized.
[image]

### Result of inclusion of any library:
[image]

### Please fix the internal functions recognition, as i am unsure now if the libraries which seem to include the internal functions even work or not.

zunhu, 2019-08-30:
Hi, may I know how can repro this issue? is uninstall all library?

[image]

beeb, 2019-09-30:
Can confirm this also happens on my end.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay