microsoft/vscode-arduino #889: Incorrect Squiggles on official MXChip samples

rido-min, 2019-09-11:
VSCode extension for Arduino shows Squiggles for almost every official MXChip sample.

[image]

or the next example

[image]

Note that in both cases the code compiles. Seeing the squiggles is really annoying and hurts the developer experience.

xinyiz1226, 2019-09-11:
Thank you for reporting this. We will take a look.

dilin-MS, 2019-10-29:
Hi @rido-min this bug is resolved in the latest Azure IoT Devce Workbench v0.10.16. Please take a look. Thanks for your participation! :)

rido-min, 2019-10-29:
Thank you very much !! I hope this small improvement will help developers who learn Azure IoT with MXChip based boards.

rido-min, 2019-10-29:
I still see squiggles like this:

[image]

dilin-MS, 2019-10-29:
Hi, are you using Visual Studio 2019 on your host computer? I just tested and found the issue when my computer have VS 2019 installed and the issue not occurs when I have VS 2017 on my computer.
Could you please try add `"compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe"` in c_cpp_properties.json file and see if the squiggles disappear? Let's see what's the root cause.

rido-min, 2019-10-31:
Yes, I have VS2019 installed. Once I changed the `compilerPath` the squiggles disappeared.

closed 2019-11-01 by dilin-MS

rido-min, 2019-11-01:
@dilin-MS may I assume the latest version will configure the right `compilerPath`? otherwise we should reopen this issue.

dilin-MS, 2019-11-03:
Hi, @rido-min . Yes, we will fix this issue in next workbench release. I open an [issue](https://github.com/microsoft/vscode-iot-workbench/issues/800) to track this.

dilin-MS, 2019-11-25:
Hi @rido-min . We have fixed the issue in the latest v0.10.17 Azure IoT Device Workbench. You can try the latest bit to confirm that. Thanks for your active participation!