microsoft/vscode-arduino #890: Double click on a warning do not navigate to the affected line

rido-min, 2019-09-11:
When there is any warning, VSCode must help by finding the offending line after clicking the warning message. This is true for C# and JS, and I would like to have a similar experience for C/Arduino projects.

xinyiz1226, 2019-09-11:
Can you provide repro steps? I played around and looks like clicking warning messages do navigate me to the corresponding line.
BTW, this functionality is provided by this extension: https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools. can you check if it is installed already?

rido-min, 2019-09-11:
I have the extension installed, and I can navigate to the offending line from the `Problems` tab, however when the build fails, or produces warnings, those are shown in the `output` tab but not in the problems tab making it difficult to locate the offending line.

[image]

henxifeng, 2019-09-24:
Please change your "intelliSenseEngine" setting to "Default" instead of "Tag Parser" in the file .vscode/settings.json. You can refer to C/C++ extension issue: https://github.com/microsoft/vscode-cpptools/issues/2164. The errors in **Output** tab when build fails will show in **Problems** tab like below.
[image]

rido-min, 2019-09-24:
Hi @hellyzh I checked my settings and I'm using `Intelli Sense Engine:Default`, and I cannot see the errors or warnings in the Problems tab after a build failure

xinyiz1226, 2019-09-25:
Can you paste content in your c_cpp_properties file? also, can you check which cpp extension version installed in your VS Code?

rido-min, 2019-09-27:
I'm using ms-vscode.cpptools 0.25.1

and here is the content of `c_cpp_properties.json`

```json
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\rmpablos\\AppData\\Local\\Arduino15\\packages\\AZ3166\\tools\\**",
                "C:\\Users\\rmpablos\\AppData\\Local\\Arduino15\\packages\\AZ3166\\hardware\\stm32f4\\1.9.3\\**",
                "C:\\Users\\rmpablos\\AppData\\Local\\Arduino15\\packages\\AZ3166\\hardware\\stm32f4\\1.9.9\\**"
            ],
            "forcedInclude": [
                "C:\\Users\\rmpablos\\AppData\\Local\\Arduino15\\packages\\AZ3166\\hardware\\stm32f4\\1.9.3\\cores\\arduino\\Arduino.h",
                "C:\\Users\\rmpablos\\AppData\\Local\\Arduino15\\packages\\AZ3166\\hardware\\stm32f4\\1.9.9\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "msvc-x64",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Preview/VC/Tools/MSVC/14.23.28019/bin/Hostx64/x64/cl.exe",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```

hellyzh, 2019-09-27:
I can repro the issue if I don't open the file main.ino, but after I double click the ino file, Problems tab will show the error. And please double confirm the settings.json have following configurations. You can also set intelliSenseEngineFallback as disabled explicitly like below.
{
    "C_Cpp.intelliSenseEngine": "Default",
    "C_Cpp.intelliSenseEngineFallback": "Disabled"
}

hellyzh, 2019-10-18:
Hi @rido-min , have you fixed you issue? If yes, I will close it.

rido-min, 2019-10-18:
@hellyzh I have the same settings, but if the file is not opened the problems tab does not show any issues. From the output Window I can see the file and line where the error is produced, but I cannot navigate to the exact location.

hellyzh, 2019-10-21:
@rido-min, yes, the file need to be opened. This is how the extension detect the "Problem". Please refer to the issue "[ErrorSquiggles only appears in open files #3207](https://github.com/microsoft/vscode-cpptools/issues/3207)".

closed 2019-10-28 by hellyzh