microsoft/vscode-arduino #891: Debug messages being printed to output - Arduino IDE 1.8.10

ameer1234567890, 2019-09-15:
With the release of Arduino IDE 1.8.10, debug messages are being printed to VSCode's output. This is what I get for a normal Verify:
```
[Starting] Verify sketch - WardrobeDoorAlarm.ino
Please see the build logs in Output path: d:\Ameer\Working\WardrobeDoorAlarm\.vscode\build
Set log4j store directory C:\Users\Ameer\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0.118856 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@af3868
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@6abbd0]...
DEBUG StatusLogger Reconfiguration started for context[name=af3868] at URI null (org.apache.logging.log4j.core.LoggerContext@6abbd0) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@591ac8
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.jsn] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.xml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.yaml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.json] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.json] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.json] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.jsn] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-test.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.jsn] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-test.jsn] using ClassLoader.getSystemResource().
[... 337 lines cut ...]
TRACE StatusLogger Unregistering 2 MBeans: [org.apache.logging.log4j2:type=af3868,component=Appenders,name=RollingFile, org.apache.logging.log4j2:type=af3868,component=Appenders,name=Console]
TRACE StatusLogger Unregistering but no MBeans found matching 'org.apache.logging.log4j2:type=af3868,component=AsyncAppenders,name=*'
TRACE StatusLogger Unregistering but no MBeans found matching 'org.apache.logging.log4j2:type=af3868,component=AsyncLoggerRingBuffer'
TRACE StatusLogger Unregistering but no MBeans found matching 'org.apache.logging.log4j2:type=af3868,component=Loggers,name=*,subtype=RingBuffer'
TRACE StatusLogger Stopping XmlConfiguration[location=jar:file:/C:/Program%20Files%20(x86)/Arduino/lib/pde.jar!/log4j2.xml]...
TRACE StatusLogger XmlConfiguration notified 2 ReliabilityStrategies that config will be stopped.
TRACE StatusLogger XmlConfiguration stopping 1 LoggerConfigs.
TRACE StatusLogger XmlConfiguration stopping root LoggerConfig.
TRACE StatusLogger XmlConfiguration notifying ReliabilityStrategies that appenders will be stopped.
TRACE StatusLogger XmlConfiguration stopping remaining Appenders.
[... cut ...]
DEBUG StatusLogger Appender Console stopped with status true
TRACE StatusLogger XmlConfiguration stopped 2 remaining Appenders.
TRACE StatusLogger XmlConfiguration cleaning Appenders from 2 LoggerConfigs.
DEBUG StatusLogger Stopped XmlConfiguration[location=jar:file:/C:/Program%20Files%20(x86)/Arduino/lib/pde.jar!/log4j2.xml] OK
DEBUG StatusLogger Stopped LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@6abbd0] with status true
[Done] Finished verify sketch - WardrobeDoorAlarm.ino
```

zunhu, 2019-09-16:
Hi, thanks for finding.
I can repro same status, since Arduino 1.8.10 release 3 days ago. I will report to extension team.

zunhu, 2019-09-16:
Since it is not a blocking issue, it won't be fix soon but in tracking.

closed 2019-09-17 by zunhu

marcelo-lara, 2019-10-02:
Is there a way to get rid of these messages?
(where to configure this?)

mouridis, 2019-10-25:
This is not a priority issue? I cannot imagine a new user that tries the extension sticking with it. You need 1 minute of scrolling to find the 2 lines of output that matter between 430 (!!!) lines of garbage output. It's unusable.

I think there should at least be a note on the "Prerequisites" section of the extension page in GitHub and Extensions Marketplace to warn people about Arduino 1.8.10 (similar to the warning about 1.8.7) until this is resolved.

xrstf, 2019-10-28:
It seems the excessive debug logging is caused by running Java with `-DDEBUG=true`. Adding `-DDEBUG=false` to the `C:\Program Files (x86)\Arduino\arduino_debug.l4j.ini` fixes it for me.

SAMXPS, 2019-11-11:
> It seems the excessive debug logging is caused by running Java with `-DDEBUG=true`. Adding `-DDEBUG=false` to the `C:\Program Files (x86)\Arduino\arduino_debug.l4j.ini` fixes it for me.

Thanks! It fixed the extensive DEBUG output console! 😉

MagTun, 2019-11-17:
Adding `-DDEBUG=false`  to `C:\Program Files (x86)\Arduino\arduino_debug.l4j.ini`  didn't work for me.

hoekbrwr, 2019-11-19:
It worked for me.@MagTun Are you sure you did put it in the right file?

MagTun, 2019-11-19:
I retried today and it worked. Not sure what happen.

dbonnell, 2019-12-04:
Thank you @xrstf, it's been driving me nuts having to scroll through hundreds of lines of crud to find the real output!

I didn't pay close enough attention first time and made the change to the wrong INI file.  There are two of them and the extra line must go in the *_debug.l4j.ini for it to work.

terrillmoore, 2019-12-06:
@zunhu On my system (Win10 Pro), I had to use file explorer>properties, and take ownership of the file in order to edit it. (I could have used an editor at an elevated prompt, perhaps, but taking ownership seemed less of a blunt instrument.) This also is not really something most Arduino users will know how to do, especially on locked-down/corporate systems. I strongly suggest treating this as a blocking issue and prioritize correction.

marcelo-lara, 2019-12-07:
Thanks @xrstf, works for me!

MatthewSteeves, 2020-01-11:
@terrillmoore 
I hit same issue. However, one of the easiest solutions is to just edit it in VSCode. It's smart/kind enough to notice that it failed to save, and ask if you want to re-try the save as Admin. It's a feature I really appreciate as the OS becomes more locked down!

MatthewSteeves, 2020-01-11:
Hey @zunhu,
I know you linked this issue to the backlog, but why is it closed if it's not resolved?
Thanks,
Matt

makeway4pK, 2020-02-29:
@zunhu might be just an intern

makeway4pK, 2020-02-29:
Thanks a lot, @xrstf ! That info might be trivial to you, but a great time saver for me 😁

caysira, 2020-03-25:
Adding -DDEBUG=false to C:\Program Files (x86)\Arduino\arduino_debug.l4j.ini worked for me! Thanks a lot.

joaomcarlos, 2020-03-26:
Let me add that there is 2 `.ini` files and that this only works for the one with `debug` in it.

I couldnt figure out why it wasnt working, and then realized I was editing the wrong file.

hakimcoder, 2020-04-04:
> It seems the excessive debug logging is caused by running Java with `-DDEBUG=true`. Adding `-DDEBUG=false` to the `C:\Program Files (x86)\Arduino\arduino_debug.l4j.ini` fixes it for me.

it's working for me! thanks for sharing your solution :-)

gautamdamodaran, 2020-04-12:
It worked for me by changing the INI file and after re-starting the VS code !

potatowave, 2020-05-31:
Is there a solution for those on macos?

Downgrading to 1.8.9 isn't an option in Catalina since 32 bit is no longer supported.

vidia, 2020-07-30:
Is there a solution for this that can be added to the extension? It should not be necessary to edit the config files of the main Arduino install to get this extension to work. Related to that, why is this extension running the debug version of Arduino where this would apply?

krassimir-nedelchev, 2021-02-13:
try to add arduino.loglevel setting to local project settings.json file (.vscode folder):
{
    "files.associations": {
        "glcdfont.c": "cpp"
    },
    **"arduino.logLevel": "info"**
}

GF-Huang, 2021-02-21:
Still not fixed, this issue should re-open!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

GF-Huang, 2021-02-21:
> try to set arduino.loglevel in local project settings.json file (.vscode folder):
> {
> "files.associations": {
> "glcdfont.c": "cpp"
> },
> **"arduino.logLevel": "info"**
> }
> 
> or change it in Ardiono extension settings.json file from "verbose" to "info".

Not work.

krassimir-nedelchev, 2021-02-21:
Did you change this setting?
[image: image.png]

krassimir-nedelchev, 2021-02-21:
Did you change this setting? 
[image]

GF-Huang, 2021-02-21:
I think this is the same:

[image]

krassimir-nedelchev, 2021-02-21:
Yes, this is global setting.

GF-Huang, 2021-02-21:
Wasn't `verbose` output more than `info`? Why `verbose`?