microsoft/vscode-arduino #892: default include config

tobiges, 2019-09-16:
Would it be possible to add a default c_cpp_properties.json so that all libraries which are included by the Arduino compiler are included in the IntelliSense config?

xinyiz1226, 2019-09-17:
Thank you for reporting this. Do you meet with intellisense issue with Devkit samples? We have another [issue](https://github.com/microsoft/vscode-arduino/issues/889) tracking this. Please let me know if you are working on different projects.

dmikhalsky, 2019-09-19:
> Thank you for reporting this. Do you meet with intellisense issue with Devkit samples? We have another [issue](https://github.com/microsoft/vscode-arduino/issues/889) tracking this. Please let me know if you are working on different projects.

I think it's different issue. Desired behaviour is the following:
- autoadd arduino library path to the "include.path" section of c_cpp_properties.json file

This should not be a problem as currently some info is already being auto-added, eg a path to toolchain

tobiges, 2019-09-22:
@xinyiz1226 It would be great if all sources here would be automatically included: [Arduino build process](https://github.com/arduino/Arduino/wiki/Build-Process#compilation)
It's very annoying to insert all the paths to the required libraries into the c_cpp_properties.json

sslupsky, 2019-10-10:
Hi,

I was looking recently at the vector-of-bool CMake Tools plugin (which appears to be a microsoft plugin now, see https://github.com/microsoft/vscode-cmake-tools).  This tool has an interesting feature that integrates with the c/c++ intellisense engine to do what the OP suggested here.  That is, CMake Tools will auto configure c_cpp_properties.json with the includes and defines from a cmake project.

Perhaps this functionality could be integrated into the Arduino plugin?

hellyzh, 2019-10-31:
We actually already have arduino library path autoadded to the "include.path" section of c_cpp_properties.json file. And looks like that is not enough, we should also include "compilerPath" to it.
In the tutorial [Using C++ and WSL in VS Code](https://code.visualstudio.com/docs/cpp/config-wsl), it states "The Compiler path setting is the most important setting".  You can refer the [issue](https://github.com/microsoft/vscode-arduino/issues/889)  try add "compilerPath" below in c_cpp_properties.json and see if everything works:
````bash
# Windows: 
"compilerPath": "C:\\Users\\<your-username>\\AppData\\Local\\Arduino15\\packages\\AZ3166\\tools\\arm-none-eabi-gcc\\5_4-2016q3\\bin\\arm-none-eabi-g++"
# Ubuntu: 
"compilerPath": "/home/<your-username>/.arduino15/packages/AZ3166/tools/arm-none-eabi-gcc/5_4-2016q3/bin/arm-none-eabi-g++"
# Mac: 
"compilerPath": "/Users/<your-username>/Library/Arduino15/packages/AZ3166/tools/arm-none-eabi-gcc/5_4-2016q3/bin/arm-none-eabi-g++"
````

sslupsky, 2019-10-31:
Hi @hellyzh I have already set my compiler path.  The other items that need to be set are the DEFINES.  These are just as important because there is a lot of condition compilation that goes on based on the defines.

I think what the community is looking for is something that extracts the library paths from the arduino compile statements.  You would need to parse the build output to do so.

There is a problem with the way the plugin determines the include paths.  As reported in issue https://github.com/microsoft/vscode-arduino/issues/850, this causes some major problems because the user cannot disable the paths that you auto add.  The auto update adds two recursive paths, one to the core root and another to the tools root. Unfortunately, those two recursive paths have a tremendous number of conflicts which leads to countless time running down squiggles that are not indicative of problems.  IntelliSense will almost always pick the wrong path and the wrong include when there are multiple includes in the recursive path(s).

Please remove the recursive directory paths provided by this feature and replace them with multiple non recursive paths. Please also give us the option to turn this feature off until at least this feature is working properly. The option to turn off a path could be granular enough to turn on or off specific paths the plugin thinks would be helpful.

elektronikworkshop, 2020-02-04:
Yeah, compiler output parsing is the way to go. This way the settings carry always valid paths and preprocessor flags even when boards are switched.

elektronikworkshop, 2020-02-05:
Hi to everyone who is suffering from this!

I was so annoyed, that I began to work on an auto-configuration mechanism, which parses the build output from Arduino and creates a `c_cpp_properties.json` from it. The goal is that no one wants likely ever edit this file manually again - except for very weird setups for which I will add a configuration flag/option that will keep the extension from touching the file.

I coded some proof of concept already but I haven't committed anything yet. But I began documenting the project [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md).

Anyone who wants to keep me motivated can chip in some :beer:-money - the donation button can be found in the link above.

Other support is appreciated and comprises
* people with Windows and OSX setups to test the cross platform compatibility
* the original maintainers could help me with CI and how to build test-releases

The progress can be followed within the same file. The [repo](https://github.com/elektronikworkshop/vscode-arduino) is set to this feature branch by default.

adiazulay, 2020-12-16:
Tracking this with issue #438

closed 2020-12-16 by adiazulay