microsoft/vscode-arduino #894: Can't install properly

mnebesny, 2019-09-19:
When I try to install, not all of the commands are there. It is notably missing Arduino: Initialize and a few others. How do I fix this?

[image]

zunhu, 2019-09-20:
hi, looks your screenshot is for settings, for commands you can try to "command palette" (ctrl+shift+P) to open command window. it will list all commands of Arduino.
[image]

mnebesny, 2019-09-21:
Thanks for the help, but the same commands are still missing. When I try to manually enter commands like Arduino: Initialize or Arduino: Examples it just opens the settings.

[image]

zunhu, 2019-09-23:
hi, can you please type "Arduino" in command box then it will be filter all commands of Arduino.

Otherwise, I can't repro this issue on my side, maybe you can try to uninstall\ install Arduino Extension and Arduino IDE and try again.

closed 2019-10-09 by zunhu