microsoft/vscode-arduino #898: Custom programmers are not listed (nrf5 based boards)

dmikhalsky, 2019-09-23:
[Arduino Core for Nordic Semiconductor nRF5 based boards](https://github.com/sandeepmistry/arduino-nRF5) adds a number of additional programmers that are listed in Arduino IDE below the standard ones

[image]<br /><a target='_blank' href='https://ru.imgbb.com/'>image hosting</a><br />

However, VS Code Arduino extension DOES NOT list them in "Select programmer" dialog

[image]

dmikhalsky, 2019-09-26:
It's not a "feature request". It's a blocking issue since its impossible to program nrf5-based boards from within VS Code

sstaub, 2019-10-04:
The same problem exist on other Arduino boards, missing are EDBG programmers and others.
In the moment the extension is complete useless.
[image]

a7hybnj2, 2019-10-11:
I use minicore and tinycore which have some chips that need their own programmer selection like "USBasp (ATTinyCore)" and those don't show up. This extension should pull the same list of programmers that the arduino IDE pulls.

barbudor, 2019-12-14:
I have the same issue with Digistump
vscode-arduino can read the board.txt to build correctly but cannot used Digistump specific programmer `micronucleus`

Current work around is to have a flash-it.cmd/flash-it.sh that can be called from the terminal.

a7hybnj2, 2020-04-26:
> Current work around is to have a flash-it.cmd/flash-it.sh that can be called from the terminal.

Can you describe what flash-it.* does? Is it project specific? Can it be linked to the 'upload' button so you don't have to run it manually?

barbudor, 2020-04-26:
It calls avrdude with the right patameters including the hex file.
So it is programmer and project dependent
But all things that Arduino IDE can do transparently from boards.txt and programmers.txt

antunjm, 2020-05-14:
Hi, I have the same problem.
I did not found other solutions than making a shell script to upload the .hex by micronucleus.
I hope the guys can implement this in the future.
Anyway, it still a great extension.

Best regards.

hairlesshobo, 2020-06-03:
I had this same problem using a home built programmer that I prefer to use instead of USB upload with the Arduino bootloader. I submitted PR #767 for this exact issue over a year ago and it has been sitting open and pending review this entire time. I just added another comment to hopefully get a maintainer to review the PR. Maybe if you guys add comments to it as well they will consider reviewing the thing.

[image]

[image]

TheLinthus, 2020-06-21:
Just had the similar problem with jtag2updi programmer.

#767 is needing some approve. The solution seems to work pretty well.

Gurkengewuerz, 2020-10-14:
Any updates on this? Really need this!