microsoft/vscode-arduino #900: Build process does not use stable Ubuntu

PeterWone, 2019-09-25:
Travis is rejecting my PR but only for Ubuntu (it passes for both Win32 and Darwin) so I built an Ubuntu VM and tooled it up until I could run the tests in VS Code.

The problem is not with VS Code or vscode-arduino, it's with the Arduino IDE which has problems on the latest Ubuntu.

I repeated the exercise with Ubuntu Stable (18.04.2) and had no problems with Arduino IDE so I finished tooling up and found that my version of vscode-arduino passes all tests.

# Fix your Travis config to use Ubuntu 18.04.2

zunhu, 2019-09-26:
hi, yes please use ubuntu 18.04.2. not support for latest version of ubuntu yet.

dooriya, 2019-09-26:
@PeterWone  As we discussed in #869, I think this issue has already been fixed.  So close it here.
Please feel free to re-open it if this issue arises again.

closed 2019-09-26 by dooriya