microsoft/vscode-arduino #901: Is ip port supported for yun?

ssduracell, 2019-09-26:
i can only select serial ports is there a way i can connect with an ip? 
needed for dragino etc.

thanks!

zunhu, 2019-09-26:
hi, select serial port is to connect with boards. 
May I know more details of your problem?

ssduracell, 2019-09-26:
i want to connect to my dragino (arduino yun + arduino uno) gateway over my network but my arduino ide doesnt find lg01

skoky, 2019-12-21:
I wish to connect to YUN with IP console as well

robotdad, 2020-10-30:
I realize serial isn't the right label here but it seems worth considering within context of discussions there.

gcampbell-msft, 2023-02-16:
@skoky @ssduracell We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature! The new Serial Monitor extension supports a TCP mode, maybe this would fit your needs?

Let us know if it solves your issue or if the issue still persists!

closed 2023-04-09 by github-actions[bot]