microsoft/vscode-arduino #902: Log4J debug spew during Arduino build

johnseghersmsft, 2019-09-26:
I'm getting what appears to be TRACE-level debug spew from Log4j when I build in VS Code using Arduino 1.8.10. The start of the build shows:
 ```
[Starting] Verify sketch - DmxInspector.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Verify could be slow. See README.
Set log4j store directory C:\Users\johnse\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0.050897 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@af3868
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@b107fd]...
DEBUG StatusLogger Reconfiguration started for context[name=af3868] at URI null (org.apache.logging.log4j.core.LoggerContext@b107fd) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@1bffc17
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using ClassLoader.getSystemResource().

**** Goes for 395 lines ending in this bit before the actual build results:

TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
TRACE StatusLogger Using DummyNanoClock for nanosecond timestamps.
DEBUG StatusLogger Reconfiguration complete for context[name=af3868] at URI jar:file:/C:/arduino/lib/pde.jar!/log4j2.xml (org.apache.logging.log4j.core.LoggerContext@b107fd) with optional ClassLoader: null
DEBUG StatusLogger Shutdown hook enabled. Registering a new one.
DEBUG StatusLogger LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@b107fd] started OK.
Preparing boards...
Verifying...
Sketch uses 10982 bytes (4%) of program storage space. Maximum is 253952 bytes.
Global variables use 2251 bytes (27%) of dynamic memory, leaving 5941 bytes for local variables. Maximum is 8192 bytes.
```
It then pics up again after the build results for another 30 lines.
Building in Arduino 1.8.10 IDE doesn't show any of this spew.
Uninstalling Arduino 1.8.10 and installing Arduino 1.8.9 gets rid of this debug spew.

zunhu, 2019-09-27:
It is same issue with #891 due to Arduino IDE upgraded to 1.8.10.

Arduino-extension dev team know this issue and tracking, since it is not blocking issue they will not fix it soon.

johnseghersmsft, 2019-09-27:
Ah, thank you for pointing to the existing issue. Since it was closed it didn't show up in my searches.

closed 2019-09-27 by johnseghersmsft