microsoft/vscode-arduino #903: In settings.json, use "Default" instead of "Tag Parser" for IntelliSense Engine

hellyzh, 2019-09-27:
In MXChip IoT DevKit samples, the setting of C_Cpp.intelliSenseEngine is "Tag Parser", change it to "Default" as the IntelliSense Engine

zunhu, 2019-09-29:
dev is working on that.

hellyzh, 2019-10-18:
The "Default" IntelliSense engine will be the eventual replacement for the "Tag Parser":
[What is the difference between includePath and browse.path?#](https://code.visualstudio.com/docs/cpp/faq-cpp)

https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference
browse The set of properties used when "C_Cpp.intelliSenseEngine" is set to "Tag Parser" (also referred to as "fuzzy" IntelliSense, or the "browse" engine). These properties are also used by the Go To Definition/Declaration features, or when the "Default" IntelliSense engine is unable to resolve the #includes in your source files.

hellyzh, 2019-11-04:
Removed "Tag Parser"

closed 2019-11-04 by hellyzh