microsoft/vscode-arduino #905: Plugin linting renames Processing (.pde) sketches into .ino files

proxycase, 2019-09-30:
On save, if plugin is enabled, this plugin will change the .pde file extension into an .ino. Didn't notice it until my entire repo had been changed into .ino files for some reason. I also have an Arduino project within the same repo, but separated by folders for Arduino sketches and Processing sketches.

zunhu, 2019-09-30:
hi, it is by design to support .pde in 0.2.26 release.

https://github.com/microsoft/vscode-arduino/blob/v0.2.26/CHANGELOG.md

proxycase, 2019-09-30:
I saw that -- that's great, but its internal linting automatically switches the filename on you to .ino instead of .pde so that all of a sudden it doesn't open in the official Processing app unless you change them all back. I'll see if I get a free moment to fix then file a push req.

swortain, 2019-10-07:
yes, I met the same problem....
I tried to set the "arduino.allowPDEFiletype": false But it seemed that it doesn't recognize by settings.json and doesn't work....

hellyzh, 2020-01-09:
If use the latest version, the setting `"arduino.allowPDEFiletype": false` should work, and it is disabled by default.

hellyzh, 2020-02-19:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-02-19 by hellyzh