microsoft/vscode-arduino #909: Cannot have multiple VSCode windows connecting to separate serial ports

SaintWacko, 2019-10-11:
If you have multiple VS Code windows open in a single project you cannot select different COM ports on each window. Changing the COM port on one window will cause the other one to change as well. I'm sure it's related to how the extension is storing that data between sessions, but hopefully there's another way to do it. It makes working on multi-board projects difficult.

SaintWacko, 2019-10-11:
On further review, it's not at all designed to have a single project/folder with multiple sketches in it. I'm not sure if this is a bug that should be fixed or if I'm just using it wrong.

zunhu, 2019-10-12:
hi @SaintWacko , it's just support one serial ports even another VSCode opened. it is by design.

closed 2019-10-17 by zunhu