microsoft/vscode-arduino #918: Unable to start serial monitor

PeterWone, 2019-11-07:
Same symptoms as #851 but with 0.2.27 on VS Code 1.40.0

jeff-winn, 2019-11-08:
I just ran into the exact same behavior. Upgraded to VS Code 1.40.0 and my ability to connect to the serial monitor has been broken.

If you need to use the Serial monitor, you can always use the Arduino IDE instead as a workaround for that as well.

basaravia, 2019-11-08:
Can't use <Select Serial Port> for ESP32 and ESP32 Lora TTGO v1. So I modified `arduino.json` it worked to upload but when try to open **Serial Monitor** appears: 

> Failed to open serial port /dev/cu.SLAB_USBtoUART due to error: + TypeError: Cannot read property 'close' of null

seems #851 related. Aditionaly installed `vscode-arduino-0.2.26.vsix` version doesn't work

secmask, 2019-11-08:
same error here when upgrade to 1.4.0

dooriya, 2019-11-08:
@PeterWone  Thank you for reporting this issue.

We're currently investigating the root cause of this issue, probably because of the Electron and Node version upgrade in latest VS Code 1.40.

mathertel, 2019-11-08:
reverting to latest-1 solves the problem (for now)
Download links available at: https://code.visualstudio.com/updates/v1_39

dooriya, 2019-11-13:
We're currently working on a new release containing the fix to this issue, will update you once we release goes public.

PeterWone, 2019-11-19:
Thanks for your efforts, I have tested vscode-arduino 0.2.28 with VSCode 1.40.1 and full compatibility has been restored.

closed 2019-11-19 by PeterWone

aubergene, 2019-12-06:
I'm still having this issue. I get the following output when opening, but no values appear

```
[Starting] Opening the serial port - /dev/cu.usbmodem14101
[Info] Opened the serial port - /dev/cu.usbmodem14101
```

```
VS Code Arduino 0.2.28
```

```
Version: 1.40.2
Commit: f359dd69833dd8800b54d458f6d37ab7c78df520
Date: 2019-11-25T14:52:45.129Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Darwin x64 18.7.0
```

PeterWone, 2019-12-09:
@aubergene The problem does not occur for 
```
Version: 1.40.2 (user setup)
Commit: f359dd69833dd8800b54d458f6d37ab7c78df520
Date: 2019-11-25T14:54:45.096Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.18363
```
Occasionally I've had problems after updating Node that were fixed by a reboot. If you reboot and you can still reproduce the problem then maybe you have versions of Node and paths are causing the wrong version to be used, but this seems unlikely given that 12.4.0 is being reported. 

If all else fails then create a new issue and include the facts that it is platform specific and you have already tried these remedies.

me6ran, 2020-03-22:
having the same issue with Version: 1.43.1

Seefer, 2020-10-20:
> Same symptoms as #851 but with 0.2.27 on VS Code 1.40.0
> 
> Same workaround -- manually edit `arduino.json` but this leaves you without Serial monitor. Better to revert VS Code to 1.39.2 and disable auto-update pending an update to this extension.
> 
> Looking at the related VSCode issue pertaining to the 1.36 breakage,
> 
> > OK. This explains it. The problem is that the Arduino extension uses native node modules (which is not official supported right now in VS Code) and the node version of VS Code has changed. This basically means that the Arduino folks need to ship a new version of their extension.
> 
> It's the same problem with the same fix.

Same here. I entered port details manually in arduino.json and uploading the Get Started code to the device seemed to work fine. The next step in Get Started Docs explaining to open serial monitoring fails completely though with this unhelpful error:

```
Failed to open serial port COM7 due to error: + Error: \\?\c:\Users\seefe\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application.
\\?\c:\Users\seefe\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node
```

Out of curiosity I fired up Windows Terminal and opened Azure Cloud Shell and used the `az iot hub monitor-events` to point at my iothub and it seems Telemtery is successfuly being sent. Can anyone clarify whether this is a VS Code bug, an Arduino extension bug or both. It's making for an appalling user experience trying to follow the **IoT DevKit Getting Started** tutorial on Microsoft Docs site that's for sure.