microsoft/vscode-arduino #919: unable to use serial port

Puneetkumarpal, 2019-11-08:
Issue Type: <b>Bug</b>

after new update visual code is unable to open serial port.

Extension version: 0.2.27
VS Code version: Code 1.40.0 (86405ea23e3937316009fc27c9361deee66ffbf5, 2019-11-06T17:01:56.781Z)
OS version: Windows_NT ia32 6.1.7601

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Celeron(R) CPU 1037U @ 1.80GHz (2 x 1795)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: disabled_off<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: unavailable_off<br>skia_renderer: disabled_off<br>surface_control: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off<br>webgl: enabled<br>webgl2: unavailable_off|
|Load (avg)|undefined|
|Memory (System)|3.39GB (1.13GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

mathertel, 2019-11-08:
duplicate to https://github.com/microsoft/vscode-arduino/issues/918

dooriya, 2019-11-15:
The serial port issue has been fixed in the latest Arduino extension release (0.2.28).
Please install the latest Arduino 0.2.28 from marketplace https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino to verify if this is resolved on your environment.

Puneetkumarpal, 2019-11-16:
Ok thank you sir

dooriya, 2019-11-20:
Thanks you all for the feedback,
Since this issue is resolved in the latest 0.2.28 release, I gonna to close it here.

closed 2019-11-20 by dooriya