microsoft/vscode-arduino #920: Unable to select serial port

zeier, 2019-11-08:
Issue Type: <b>Bug</b>

I have set up the development environment as described here: https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-arduino-iot-devkit-az3166-get-started#prepare-the-development-environment

With Tera Term, I'm able to communicate with the IoT DevKit over the corresponding serial port (COM5).

However, in VS Code, I cannot select a serial port (nothing happens when I click on "Select Serial Port").

Extension version: 0.2.27
VS Code version: Code 1.40.0 (86405ea23e3937316009fc27c9361deee66ffbf5, 2019-11-06T17:02:13.381Z)
OS version: Windows_NT x64 10.0.18362

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-8350U CPU @ 1.70GHz (8 x 1896)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: enabled<br>rasterization: enabled<br>skia_renderer: disabled_off<br>surface_control: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.93GB (2.77GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

mathertel, 2019-11-08:
duplicate to https://github.com/microsoft/vscode-arduino/issues/918

dooriya, 2019-11-13:
We're currently working on a new release containing the fix to this issue, will update you once we release is go public.

dooriya, 2019-11-15:
The serial port issue has been fixed in the latest Arduino extension release (0.2.28).
Please install the latest Arduino 0.2.28 from marketplace https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino to verify if this is resolved on your environment.

dooriya, 2019-11-20:
Thanks you all for the feedback,
Since this issue is resolved in the latest 0.2.28 release, I gonna to close it here.

closed 2019-11-20 by dooriya

zeier, 2019-11-20:
Verified with version 0.2.28 that the issue does not occur anymore on my environment.

simonl65, 2020-09-20:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```

schimini, 2020-09-21:
This is still an issue (v1.49.1)

simonl65, 2020-09-21:
Indeed it is 👎