microsoft/vscode-arduino #922: Select A Serial Port doesn't work

ryancor, 2019-11-09:
Issue Type: <b>Bug</b>

Im on Mac, and everytime i try to select a serial port at the bottom, nothing shows up.

Extension version: 0.2.27
VS Code version: Code 1.40.0 (86405ea23e3937316009fc27c9361deee66ffbf5, 2019-11-06T17:09:34.601Z)
OS version: Darwin x64 18.7.0

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz (8 x 2500)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off<br>surface_control: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|2, 2, 2|
|Memory (System)|16.00GB (7.83GB free)|
|Process Argv|-psn_0_7046840|
|Screen Reader|no|
|VM|0%|
</details>

AldebaranPrimo, 2019-11-11:
The same thing happens in windows

Extension version: 0.2.27
VS Code version: Code 1.40.0 
OS version: Windows 10 pro 1903 (build 19018.1)

acrandal, 2019-11-11:
I have the same thing on a Linux Mint (Debian/Ubuntu) system. The ArduinoIDE packages work, but VS Code doesn't let me select a serial port. This seems to be the same thing as issues #596, #664 , #816, and/or #851. The release of 0.2.27 fixed this problem for some people, but not everyone, it seems. 

If I edit the .vscode/arduino.json file directly and add the line for the serial port path in my case to:
"port": "/dev/ttyUSB0",

The VS Code interface picks up the configuration change and can program the Arduino Uno. When I try to bring up the serial console I get:

"Failed to open serial port COM6 due to error: + TypeError: Cannot read property 'close' of null"

vscode-arduino version: 0.2.27
vscode version: 1.40.0
Electron: 6.1.2
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 4.15.0-65-generic

I have read/write perms on the serial port and can run other tools without priv escalation to read/write just fine.

To sum up:
* The "Select Serial Port" button and menu option don't bring up a menu.
* Setting the serial port in the json file manually allows programming, but not use of the serial console.

smarpug, 2019-11-12:
I am having this problem as well on Windows Version 10.0.18362 Build 18362.
vscode-arduino version: 0.2.27
vscode version: 1.40.0

I can interact with the serial device via PuTTY, and I can also set the COM port in the arduino.json, but the button at the bottom right and the Command Palette command both do nothing as far as the UI is concerned.

stevenjoezhang, 2019-11-12:
Same issue here - Nothing shows up after clicking the `<Select Serial Port>` button.

vscode-arduino version: 0.2.27
vscode version: 1.40.0
OS version: Darwin x64 19.0.0 (macOS Catalina 10.15.1)

ThomasOdegard, 2019-11-12:
Same issue.
Arduino ext.version: 0.2.27
VScode version: 1.40.0
Electron: 6.1.2
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Win10

arduino.log showing: 

{"correlationId":"4e71062607b3432184bf6cdd6cec17b6","command":"arduino.selectSerialPort","message":"error","errorLine":"\tat Object.<anonymous> (c:/users/..../.vscode/extensions/vsciot-vscode.vscode-arduino-0.2.27/vendor/node-usb-native/lib/bindings.js:27:19)","telemetry":true,"level":"error","timestamp":"2019-11-12T08:41:12.402Z"}

Dhdrummer, 2019-11-12:
I have the same issue, clicking on a COM port name in status bar does nothing.

jrlanglois, 2019-11-12:
To be totally honest, this might be related to the Arduino IDE installation. Why people are pasting things like how much RAM they have is bizarre.

My set up is as follows:
```
Extension version: 0.2.27
VS Code version: Code 1.40.0 (86405ea23e3937316009fc27c9361deee66ffbf5, 2019-11-06T17:02:13.381Z)
OS version: Windows_NT x64 10.0.18362

Arduino IDE: 1.8.10
```

I will try an older version or two of the IDE and compare results...

jrlanglois, 2019-11-12:
Quick update: testing out [vscode 1.39.2](https://code.visualstudio.com/updates/v1_39) and the COM port selection works just fine.

acrandal, 2019-11-12:
I can confirm @jrlanglois finding that vscode 1.39.2 has a working serial port selection.

Version: 1.39.2
Commit: 6ab598523be7a800d7f3eb4d92d7ab9a66069390
Date: 2019-10-15T15:33:40.634Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Linux x64 4.15.0-66-generic

Arduino IDE: 1.8.9

dooriya, 2019-11-13:
Duplicated with #918 #927 due to VSCode engine bumped to 1.40 with Electron 6.0 update.

dooriya, 2019-11-13:
We're currently working on a new release containing the fix to this issue, will update you once we release is go public.

dooriya, 2019-11-15:
The serial port issue has been fixed in the latest Arduino extension release (0.2.28).
Please install the latest Arduino 0.2.28 from marketplace https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino to verify if this is resolved on your environment.

MikeX7, 2019-11-17:
It fixed the issue for me

acrandal, 2019-11-19:
It seems to have worked me for my systems as well.

@dooriya and team:
Thank you so much, devs! Having VS Code for Arduino work is a great tool.

smarpug, 2019-11-19:
Excellent. This solved the issue on my environment (summarized in previous comment).

jrlanglois, 2019-11-19:
+1, v0.2.28 fixes the issue for me. Thanks!

dooriya, 2019-11-20:
Thanks you all for the feedback,
Since this issue is resolved in the latest 0.2.28 release, I gonna to close it here.

closed 2019-11-20 by dooriya

Christian1984, 2020-03-11:
Unfortuanately, I still encounter the issue described above. I'm on:

Mac OSX 10.14.6
Arduino IDE 1.8.11
VSCode 1.43.0
Arduino-Extension 0.2.29

Going through the command palette didn't work either. After setting the port manually through arduino.json upload worked, but serial monitor still does not.

smarpug, 2020-03-11:
This issue was fixed earlier, but it has re-appeared for me as well.
Windows Version 10 Build 19041.113
vscode-arduino version: 0.2.29
vscode version: 1.43.0

smarpug, 2020-03-11:
I can also set the port manually through arduino.json without any problem, but serial monitor does not work.

antonu17, 2020-03-13:
Same here.
Can't select serial port via menu. If configure port via arduino.json - upload works.
MacOS 10.15.3 (19D76)
VSCode 1.43.0 (Electron: 7.1.11, V8: 7.8.279.23-electron.0)
vscode-arduino 0.2.29

rafaellucas3, 2020-03-13:
After the update I can't select a new COM port or user the serial terminal. 

Version: 1.43.0 (user setup)
Commit: 78a4c91400152c0f27ba4d363eb56d2835f9903a
Date: 2020-03-09T19:47:57.235Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18363

cyberflying, 2020-03-16:
select serial port does not work.

jacques42, 2020-03-17:
I can confirm same problem here. I manually added the serial port to arduino.json and that fixed it.
"port": "COM4"

daggilli, 2020-03-18:
I'm using VS Code 1.43.0 with the 0.2.29 Arduino extension and the 1.8.12 Arduino IDE under Ubuntu 18.04. The serial port selection is non-functional for me as well. I fixed it by setting the correct TTY device driver in my `arduino.json` to `/dev/ttyACM0` which is the default for Linux.

marcojak, 2020-03-20:
It doesn't work on VS Code 1.43.1 and Arduino extension 0.2.29

Smetanski, 2020-03-20:
> I'm using VS Code 1.43.0 with the 0.2.29 Arduino extension and the 1.8.12 Arduino IDE under Ubuntu 18.04. The serial port selection is non-functional for me as well. I fixed it by setting the correct TTY device driver in my `arduino.json` to `/dev/ttyACM0` which is the default for Linux.

What exactly you write in arduino.json file?

clarkbremer, 2020-03-20:
Same: "select serial port" does nothing.   VS Code1.43.1, Arduino Extension 0,2.29 on Windows 10 v1909.

daggilli, 2020-03-20:
> What exactly you write in arduino.json file?

```
{
    "programmer": "AVR ISP",
    "board": "arduino:avr:uno",
    "sketch": "sketch.ino",
    "port": "/dev/ttyACM0"
}: 
```

rafaellucas3, 2020-03-20:
The serial monitor doesn't work even if you add the port line on the arduino.json. Install this one:
[Visual Studio Code 1.42](https://update.code.visualstudio.com/1.42.1/win32-x64-user/stable
). After installtion stop automatic update by modifying the update mode. Go to “File” > “Preferences” > “Settings, search for update mode and change the setting to “none”.

daggilli, 2020-03-20:
I can't get the serial monitor to work even after a downgrade. After `Arduino: Open Serial Monitor` I get an editor panel but nothing ever appears on it. The serial monitor in the Arduino IDE works fine. I might as well use VS Code 1.43 and set the port in my `settings.json`..

rafaellucas3, 2020-03-20:
Is the baud rate configured correctly? I did this blog post with the 1.42:
https://rafaelsouza.net/wp-admin/post.php?post=37&action=edit
You can see it worked for me.

daggilli, 2020-03-20:
That was it. I set the default baud rate (`arduino.defaultBaudRate`) to be 9600 to match my code and it started working.

clarkbremer, 2020-03-21:
Seems there's a PR in the works:  https://github.com/microsoft/vscode-arduino/pull/996

plevec37, 2020-03-22:
Same problem,
VScode 1.42 solves the problem.
Merci

bharatraja, 2020-03-26:
I have the same issue on vs cdoe 1.43.2. Can select the serial port

clarkbremer, 2020-03-26:
Fixed with latest version of Arduino extension (0.3.0)

thinkOfaNumber, 2020-09-11:
Seems to be broken again with the latest VS Code update for windows:

```Version: 1.49.0 (user setup)
Commit: e790b931385d72cf5669fcefc51cdf65990efa5d
Date: 2020-09-10T13:22:08.892Z (15 hrs ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041```

Arduino extension (uninstalled & reinstalled): v0.3.2

Elektrik1, 2020-09-17:
Yup, still broken.

simonl65, 2020-09-20:
Broken here too.

Version: 1.49.1 (user setup)
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-16T23:27:51.792Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18363

Also, clicking on the Open Seriual Monitor gives the following error:

Failed to open serial port COM1 due to error: + Error: The module '\\?\c:\Users\xxxxxx\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 75. This version of Node.js requires NODE_MODULE_VERSION 80. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).

simonl65, 2020-09-20:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```

simonl65, 2020-09-21:
Still a problem in 1.49.1 👎 (

JanGieseler, 2020-09-22:
Same issue here:

Version: 1.49.1 (user setup)
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-16T23:27:51.792Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18362

chrs4711, 2020-09-24:
Can confirm, also not working on linux

Version: 1.49.1
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-16T23:23:33.049Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Linux x64 5.4.0-48-generic

ATHONOR-electronics, 2020-09-26:
Just downloaded VS Code / Arduino extension on Windows 10, same issue. Tried reverting the extension back to 3.0 (from 3.2) made no difference. 

Version: 1.49.2 (user setup)
Commit: e5e9e69aed6e1984f7499b7af85b3d05f9a6883a
Date: 2020-09-24T16:29:41.983Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

darcynog, 2020-09-27:
This is how I SOLVED this problem in my Ubuntu machine:

1)In  the settings.json file (reacheable through File / Preferences / Settings   and then the Workspace Settings tab)
disabled the option  “arduino.enableUSBDetection” setting it to  false.  (“arduino.enableUSBDetection” : false.  )

2)Then in the file arduino.json manually specified the device name for the USB connection in my machine:
   "port": "/dev/ttyACM0".  
    
Now I can upload sketches to the Arduino board.  

I had this idea while reading  https://medium.com/home-wireless/use-visual-studio-code-for-arduino-2d0cf4c1760b

Roberto6969, 2020-09-27:
It seems good idea. But as long I have tested on windows machine, using COMx port syntax, it doesn't work

Xinyu-Wu-0000, 2020-09-28:
Seem it is a problem about nodejs
https://github.com/microsoft/vscode/issues/106500

fidelbrea, 2020-10-01:
Downgrade. This is a solution. I fixed it by downgrading up to 1.48.2.
I hope you find this useful to fix your own situations.

DerekFoster1976, 2020-10-03:
Downgrading from 1.49 to 1.48.2 works for me to, I wish they would get a handle on this issue as it has happened quite a few times now.

DerekFoster1976, 2020-10-03:
I now have another issue. I use an MXChip which has fully supported debugging with the extension. However, when I just have the .ino file open in the workspace (no other files open) and try to debug by run>Start Debugging it does not list the Arduino environment (see screenshot1), When I open an additional file such as the settings.json file the full list of environment options is shown when I do Run>Start Debugging (screenshot2).
[image]
[image]

simonl65, 2020-10-03:
@DerekFoster1976 It might be better to open that as a new issue - bit's unlikely to get noticed in this closed issue :-(

Antiquary1820, 2020-10-03:
> This is how I SOLVED this problem in my Ubuntu machine:
>
> 1)In the settings.json file (reacheable through File / Preferences / Settings and then the Workspace Settings tab)
> disabled the option [...]

This worked for me too - thanks so much darcynog - this old guy appreciates your help

Yogsther, 2020-10-08:
> This is how I SOLVED this problem in my Ubuntu machine:
>
> 1)In the settings.json file (reacheable through File / Preferences / Settings and then the Workspace Settings tab)
> disabled the option [...]

Fixed it on windows aswell.

FaimMedia, 2020-10-17:
Having the same issue, adding the device manually to the `arduino.json` file fixes the upload for me. However serial monitor doesn't work anymore.

OS version: macOS 10.15.5
VSCode: 1.50.1
Arduino extension: v0.3.2

When I try to open the serial monitor, I get the following error. There seems to be some sort of mismatch with Node.js versioning:
```
Failed to open serial port /dev/tty.usbmodem143101 due to error: + Error:
The module '/Users/Tim/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 75. 
This version of Node.js requires NODE_MODULE_VERSION 80.
Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).
```

Augustinio, 2020-10-25:
same issue over here:
OS 10.15.7
VSCode 1.50.1
Arduino extension v0.3.2

Nothing happens when clicking on select serial port

henniespies, 2020-10-27:
same issue here
Version: 1.50.1 (user setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041
Arduino Extension: v0.3.2

Xingqishiyi, 2020-10-27:
OS Windows 7 SP1
VSCode: 1.50.1
Arduino extension: v0.3.2

Initially, nothing happens when clicking on select serial port.
I then switched to Arduino IDE and built my project there.
When I returned to Visual Studio Code, it all worked.
I was able to choose serial port and upload.
Could not line out exactly all of the steps, but now have VS Code working properly without any changes to the versions.

Hope this helps...

FrankLaterza, 2021-02-16:
Yeah, same here. This is happening on both of my computers. I tied installing the new Arduino update, but nothin.

simonl65, 2021-02-16:
I gave up and installed PlatformIO instead - it's been rock-solid for me ;-)

[2 more replies]