microsoft/vscode-arduino #924: New version of Arduino extension fail to overwrite environment variable with a OS branch

DeqingSun, 2019-11-12:
Hi, today I was testing debugging Arduino Uno with VScode on Mac. The launch.json worked in September didn't work with the current release of extension.

it used to be
###
            "miDebuggerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-gdb.exe", 
            "debugServerPath": "Z:\\Github\\dwire-debug\\dwdebug.exe",
            "debugServerArgs": "verbose,gdbserver,device usbtiny1",
            "MIMode": "gdb",  
            "osx": {
                "miDebuggerPath": "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gdb",
                "debugServerPath": "/Users/sundeqing/Documents/Github/dwire-debug/dwdebug",
            },
###

On Mac, the miDebuggerPath and debugServerPath were overwritten by the osx values in September, but now it doesn't work. I have to write it like this as a quick fix.

###
            "miDebuggerPath": "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gdb", 
            "debugServerPath": "/Users/sundeqing/Documents/Github/dwire-debug/dwdebug",
            "debugServerArgs": "verbose,gdbserver,device usbtiny1",
            "MIMode": "gdb",  
            "osx": {
                "miDebuggerPath": "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gdb",
                "debugServerPath": "/Users/sundeqing/Documents/Github/dwire-debug/dwdebug",
            },
###

I didn't get a chance to hook debugger to Arduino extension. Is the OS branching feature depreciated? Or I did it in the wrong way?

Thanks
Deqing

hellyzh, 2019-11-12:
There should be no change for the debugger of Arduino extension. Have you selected the Board Type and serial port for your device?

DeqingSun, 2019-11-13:
Hi,

I just figured out the new 1.40 VScode break it. 1.39 works fine.

hellyzh, 2019-11-18:
we have released new version arduino 0.2.28, please check if it works for VScode 1.40.1

DeqingSun, 2019-11-19:
Hi,

This works now, thanks.

closed 2019-11-19 by DeqingSun