microsoft/vscode-arduino #926: [Error] Exit with code=1 when uploading

StaDo0815, 2019-11-12:
I'm having this problem with uploading to my arduino mini pro  (I'm not using PlatformIO)
[image]

hellyzh, 2019-11-13:
Does that work before? Are you using VSCode 1.40.0? There is a bug that unable to select serial port in the version.

StaDo0815, 2019-11-13:
I know the bug. For that reason I'm using 1.39.2. It never worked before.

dooriya, 2019-11-21:
@StaDo0815 Could you provide more information for this issue? E.g. your development environment, the repro steps, and the full error message/log?

At the same time, we just released a updated version for Arduino extension, could you try installing the **latest 0.2.28** from the [marketplace](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) to check if the issue still occur with the latest 0.2.28?

StaDo0815, 2019-11-24:
Sorry for not answering. The issue has been fixed by updating VSCode as well as the Arduino extension.
Thank you for helping out.
[image]

closed 2019-11-24 by StaDo0815

X-iaowh, 2020-03-04:
I also meet the same problem, it took me so much times.Then I try to rename the name of the file with an English name instead of a Chinese name,the problem was gone.

BradBath, 2020-03-14:
I'm still having this issue. I've got the latest version of the arduino plugin for VSCode and a downgraded version of VSCode to fix a bug in the latest version where clicking Select serial port wouldn't bring anything up (version is now 1.42.1).

chrisdreid, 2021-05-06:
Would like to request that the arduino.json (all settings files) file allow for variable tokens. https://code.visualstudio.com/docs/editor/variables-reference