microsoft/vscode-arduino #927: Unable to select serial port

Dhdrummer, 2019-11-12:
Issue Type: <b>Bug</b>

After the update to 1.40.0 I tried to change serial port by clicking on it in status bar but nothing happens. Every other clickable method also does not work.

Extension version: 0.2.27
VS Code version: Code 1.40.0 (86405ea23e3937316009fc27c9361deee66ffbf5, 2019-11-06T17:02:13.381Z)
OS version: Windows_NT x64 10.0.18362

jrlanglois, 2019-11-12:
I have the same configuration and am running into the same issue.

Strangely, I had this very extension working just fine (including selecting the serial port) a couple weeks ago.

hellyzh, 2019-11-13:
Unable to select serial port #918

dooriya, 2019-11-13:
Duplicated with #918 due to VSCode engine bumped to 1.40 with Electron 6.0 update.
We're currently working on a new release containing the fix to this issue, will update you once we release is go public.

dooriya, 2019-11-15:
The serial port issue has been fixed in the latest Arduino extension release (0.2.28).
Please install the latest Arduino 0.2.28 from marketplace https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino to verify if this is resolved on your environment.

josephpage, 2019-11-19:
I confirm that the issue with Serial Port is fixed with 0.2.28, on my side too.

dooriya, 2019-11-20:
Thanks you all for the feedback,
Since this issue is resolved in the latest 0.2.28 release, I gonna to close it here.

closed 2019-11-20 by dooriya

micahalan, 2020-03-15:
I am currently having this same issue with version 1.43.0

simonl65, 2020-09-20:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```