microsoft/vscode-arduino #930: Can't install libraries from Vscode for ESP8266

gtleon, 2019-11-13:
Hello!

Please tell me I would like to create a portable version of the plugin for Arduino for vscode.

In the Arduino app, I changed the location of the Arduino system folders.

In the plugin for VSCode, I changed the folder location to the ones I needed.

Basic standard sketches are verified.

1) Can't install libraries from Vscode for ESP8266 (via Arduino distribution installed).

2) After installing libraries for ESP8266 through Arduino IDE, can't verify sketch.

Thanks in advance!

hellyzh, 2020-02-19:
Support custom libraries with portable version of Arduino #946

DrXoo, 2020-03-05:
> Support custom libraries with portable version of Arduino #946

If this is already solved then the Issue can be closed?

closed 2020-03-26 by hellyzh