microsoft/vscode-arduino #935: Arduino IDE Cannot be found if C:\ is hidden

Jackbennett, 2019-11-18:
Getting code & arduino setup for students in a school, the module fail to start with the error "Cannot find IDE"

If I manually set `arduino.path` to the default `C:\Program Files (x86)\Arduino` where I know it is installed the module works fine.

I guess something in the path validation fails to traverse c:\ looking for the IDE. If it just blindly tries to run the files it wants it will find them.

Without stepping through a debug code I can't see why [extension L:94](https://github.com/microsoft/vscode-arduino/blob/54d4ea94bbfe46d7799d27f1906a8fb9578da24f/src/extension.ts#L94) wants to give that error.

As a student I can open that registry key HKLM...install_dir, and from the vscode dev console I can get `true ` for `fs.statSync('C:\\Program Files (x86)\\Arduino').isDirectory()` and also `fs.statSync('C:\\Program Files (x86)\\Arduino\\arduino_debug.exe').isFile()`

hellyzh, 2019-11-20:
Please add arduino.path as below, if you don't have permission on c:\, please copy to the Arduino folder to a drive like below which you has permission.
````bash
"arduino.path": "D:\\temp\\Arduino"
````

hellyzh, 2019-12-10:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2019-12-10 by hellyzh