microsoft/vscode-arduino #937: Board Manager not working in Docker container

mt1n, 2019-11-22:
I am running Visual Studio Code in a docker container. When I open the Board Manager I just see a connection failed message to localhost:<random port>.
In localWebServer.js I changed fixed the port and configured docker to forward to this port. I also changed the ServerUrl to my server address.
This trick seemed to work but this is not the best solution, I guess.

hellyzh, 2020-02-21:
Mark a feature request to support Docker container. And Thanks @mt1n for the work-around.

Evander12345, 2020-04-11:
Adding an optional configuration setting for the aforementioned server address and port would allow this extension to also be used in [code-server](https://github.com/cdr/code-server). Currently using this extension in code-server results in the same error as @mt1n reported.