microsoft/vscode-arduino #938: [Feature Request] Verify/upload using current INO file

modmatic, 2019-11-23:
When developing libraries with multiple example INO files (or working in any directory with multiple INO files), it's inconvenient to have to keep updating the current sketch in the JSON config/plugin.

This is a feature request to tie the "verify" and "upload" buttons to the current INO file.

It would be more intuitive and also reduce confusion when users try to build a file that they are editing while the config is actually set to build a different file.

modmatic, 2019-11-23:
Additional ideas/alternatives:

- Option to enable/disable this feature
- Functionality added using additional buttons, if current functionality must be retained.

jhoughjr, 2020-05-25:
bump. Uploads files that aren't even open.

Strat00s, 2020-07-15:
This. How is this not implemented. It makes so much sense to upload what you are currently working on instead of some random thing you uploaded sometimes before 🤦