microsoft/vscode-arduino #939: Extension causes high cpu load

luxmorpheux, 2019-11-29:
- Issue Type: `Performance`
- Extension Name: `vscode-arduino`
- Extension Version: `0.2.28`
- OS Version: `Windows_NT x64 10.0.18362`
- VSCode version: `1.33.1`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`C:\Users\Button\vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/Microsoft/vscode/wiki/Explain:-extension-causes-high-cpu-load

hellyzh, 2019-12-16:
correct url: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

hellyzh, 2020-01-06:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-01-06 by hellyzh