microsoft/vscode-arduino #942: Saved programmer not selected

manuelmaurer, 2019-12-06:
When selecting a Programmer, the selection is saved in the arduino.json file. However, this selection is not restored when loading the project (neither after "Open Folder" nor after a restart of vscode).
It also does not matter which programmer is selected (tested with multiple entries).

silardg, 2019-12-28:
Same situation when using STM32L0 boards with ST-Link programmers.

rickoman, 2020-01-25:
Same problem here with Adafruit Feather M0 + AVR mkII

manuelmaurer, 2020-03-18:
I've tried to dig into the code, it looks like there is no routine to actually load the setting into the programmerManager. Based on how this is handled in boardManagerI wrote a quick fix. Patch attached.

[loadProgrammer.txt](https://github.com/microsoft/vscode-arduino/files/4349640/loadProgrammer.txt)

hairlesshobo, 2020-06-03:
Last year I submitted pull request #767 that addresses this, and the fact that custom programmers are not honored by the vscode arduino plugin last year. It has been sitting pending review this entire time. If you guys want to comment on that, they may finally review it and hopefully merge it in so these two issues will be addressed.

THX2112, 2020-06-11:
Same issue.

iqapps, 2020-09-15:
Anything happening here, should be an easy fix.

PR #1118 (2020-12-08): Improved handling of programmer selection
- Selected programmer is now saved to and loaded from the arduino.json file
- Arduino.json is monitored for changes, and changing file will update selected programmer & ui
- Programmer selection UI now shows both the friendly name of the programmer, as well as the arduino name
- Minor fix to deviceContexts to fire change events after all states are modified
- Layed groundwork to support querying list of programmers for the current board from arduino toolchain

Fixes #942