microsoft/vscode-arduino #945: Cannot download "https://github.com/sass/node-sass/releases/download/v4.11.0/win32-x64-79_binding.node"

iitaka1142, 2019-12-14:
When I install vscode-arduino with npm, npm reports error with message below,

```shell
Downloading binary from https://github.com/sass/node-sass/releases/download/v4.11.0/win32-x64-79_binding.node
Cannot download "https://github.com/sass/node-sass/releases/download/v4.11.0/win32-x64-79_binding.node":

HTTP error 404 Not Found
```

I could install after ```npm install node-sass``.
it download v4.13.0/win32-x64-79_binding.node

scripts/install.js should be corrected.

hellyzh, 2019-12-17:
The dependency has been updated in [PR943](https://github.com/microsoft/vscode-arduino/pull/943), you can try again.

iitaka1142, 2019-12-28:
fixed, thank you.

closed 2019-12-28 by iitaka1142