microsoft/vscode-arduino #947: Verify error: java.io.IOException: No valid code files found

umatbro, 2019-12-21:
## What is expected vs what is happening

I want to verify my Arduino file but it results with an error `java.io.IOException: No valid code files found` even with the file is empty
```cpp
void setup()
{
	
}

void loop()
{
	
}
```

## Issue details
Version of VSCode: 
```
> code --version
1.41.1
26076a4de974ead31f97692a0d32f90d735645c0
x64
```
OS: Windows 10

Steps to reproduce:
1. Open new folder `File -> Open folder...`
2. Click `Ctrl + Shift + P` and select `Arduino: Initialize`
3. Create `.ino` file and select Arduino Nano board
4.  `Ctrl + Shift + P` and select `Arduino: Verify`

The output is the following: 
```
[Starting] Verify sketch - app.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Verify could be slow. See README.
Set log4j store directory C:\Users\matur\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0.114343 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@af3868
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@1d80d2b]...
DEBUG StatusLogger Reconfiguration started for context[name=af3868] at URI null (org.apache.logging.log4j.core.LoggerContext@1d80d2b) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@7bfdc1
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using sun.misc.Launcher$AppClassLoader@af3868 class loader.
TRACE StatusLogger Trying to find [log4j2-testaf3868.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-testaf3868.json] using context class loader sun.misc.Launcher$AppClassLoader@af3868.
[... 388 lines cut ...]
DEBUG StatusLogger Shut down OutputStreamManager SYSTEM_ERR.false.false, all resources released: true
DEBUG StatusLogger Appender Console stopped with status true
TRACE StatusLogger XmlConfiguration stopped 2 remaining Appenders.
TRACE StatusLogger XmlConfiguration cleaning Appenders from 2 LoggerConfigs.
DEBUG StatusLogger Stopped XmlConfiguration[location=jar:file:/C:/Program%20Files%20(x86)/Arduino/lib/pde.jar!/log4j2.xml] OK
DEBUG StatusLogger Stopped LoggerContext[name=af3868, org.apache.logging.log4j.core.LoggerContext@1d80d2b] with status true
[Error] Exit with code=1
```

So the main error seems to be:
```
java.io.IOException: No valid code files found
	at processing.app.Sketch.listSketchFiles(Sketch.java:117)
	at processing.app.Sketch.<init>(Sketch.java:54)
	at processing.app.Base.<init>(Base.java:415)
	at processing.app.Base.main(Base.java:150)
```
But my `.vscode/arduino.json` file seems to be initialized correctly
```json
{
    "sketch": "app.ino",
    "board": "arduino:avr:nano",
    "configuration": "cpu=atmega328"
}
```

hellyzh, 2020-01-08:
The expected result should be showing `"[Done] Finished verify sketch - app.ino“`.
You can try to open `"Arduino: Examples"` like Blink, and check the verify result.

hellyzh, 2020-02-17:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-02-17 by hellyzh

c66zwzly, 2020-02-28:
I meet the same problem with you. Have you solved it ?

umatbro, 2020-02-29:
@c66zwzly Unfortunately not, I switched back to Arduino IDE

mianqi2016, 2020-06-26:
I met the same problem, and I solved it.
The reason was there is unrecognized characters(for me, that was Chinese Characters) in your files path(i.e. directory name).
If you only used "verify" on up-right corner, the problem prompt would shown as you copied here：
[image]

 but if you used "Terminal -> Run Build Task...", it would show more info of the problem, for me, the place of Chinese Characters would be shown as "??".
[image]

mianqi2016, 2020-06-26:
> 
> 
> @c66zwzly Unfortunately not, I switched back to Arduino IDE

You back too quickly.