microsoft/vscode-arduino #950: Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change.

ZBHorizon, 2019-12-30:
[image]

Ich habe Windows 10 un bekomme es einfach nicht hin den Arduino.path anzugeben

hellyzh, 2020-01-06:
Have you installed [Arduino IDE](https://www.arduino.cc/en/Main/Software)? And please add the setting of "arduino.path", if in Windows, go to Visual Studio Code, click File > Preferences > Settings, then click the Open Settings (JSON) icon in the upper-right corner of the Settings page.
add `"arduino.path": "C:\\Program Files (x86)\\Arduino"`.

closed 2020-02-17 by hellyzh

mianqi2016, 2020-05-18:
I also meet this problem, and I have installed Arduino-1.8.12 on Ubuntu, but I couldn't find the location of Arduino IDE in it, so I don't know what should be filled in the "Arduino:Path" blank.

hellyzh, 2020-05-19:
Fill like this `"arduino.path":"/home/User/Arduino1.8.2/arduino-1.8.2"` (replace "/home/User/Arduino1.8.2/arduino-1.8.2" with the path of Arduino IDE).

tariqawni, 2020-07-08:
Hi i am new on vscode and i wana ask where i should to fill "arduino.path": "C:\\Program Files (x86)\\Arduino"  ? 
go to Visual Studio Code, click File > Preferences > Settings, then click the Open Settings (JSON) icon after that in what line i should to paste this "arduino.path": "C:\\Program Files (x86)\\Arduino"

hellyzh, 2020-07-10:
[image]
Click this icon.

tariqawni, 2020-07-10:
after open settings json in wich line i should to add "arduino.path": "C:\Program Files (x86)\Arduino"  ?

hellyzh, 2020-07-13:
Any line as you like. And use double backslash in windows as `"arduino.path": "C:\\Program Files (x86)\\Arduino"`.

K-zhao, 2020-07-24:
I apologize for necroing this post, but adding that didn't seem to fix the issue, even after a restart. I'm on Arduino 1.8.13. Are there potentially any other issues that might cause the extension to not function properly? 
[image]

umatbro, 2020-08-20:
I'm struggling with the same issue. `arduino` is installed in my system, it launches Arduino IDE properly. I'm using Manjaro Linux 20.0.3. Turning VS Code on and off does not have any effect.

[image]

TonyDiana, 2020-10-30:
/snap/arduino/41

edumudu, 2020-12-30:
I have installed from snap in ubuntu 20.04 and my path that works was /snap/bin

Tejas-Rathi, 2021-04-03:
GUYS TRY USING THIS CODE 

`"arduino.path": "C:\\Program Files\\Arduino IDE"`

AFTER THIS PRESS`ctrl=shift=esc`
in that window click on visual studio code and click on end task at the bottom right
boot up visual studio code again

monsefe19, 2021-04-12:
for windows 
arduino path command path (arduino.exe) 
path
(C:\Program Files (x86)\Arduino)

Andrey-Anatolyevich, 2021-04-15:
FYI:
As of now, there are 2 of Arduino IDE available: 1.8 and 2.0 beta
Letting you know, the 2.0 beta doesn't work with VS Code plugin.
You have to use version 1.8.

I've tried to use 2.0, but no luck. After lownloading 1.8 and updating settings of VS Code (arduino.path), the plugin stopped complaining it could not find Arduino IDE.

drcrook1, 2021-04-28:
I'm having this problem as well.

my settings.json
    "arduino.commandPath": "C:\\Program Files (x86)\\Arduino",

I have validated that arduino.exe is inside this folder.  I'm on v 1.8.13

PJensen, 2021-04-29:
Installing from windows store has made it hard to find for me. Best course of action seems to be to install it without the app store.

sankayop, 2021-05-09:
> @VictorVmg I'm using Linux Manjaro - as I stated in my comment. Unfortunately any of the F1 -> Arduino does not work, all are resulting in the same error :(
> 
> **EDIT** I've found the issue. In the `arduino.path` setting I was providing a path to the `arduino` executable instead of a directory.
> I changed `/usr/bin/arduino` to `/usr/bin` and now VS Code seems to be able to detect my Arduino IDE properly :)

Thanks a lot, this solved it for me :)

NunezJr, 2021-05-19:
Still no luck with getting code to find this 

[image]

why oh why does this have to be so sensitive

Iiqbal2000, 2021-06-17:
If you install Arduino IDE using Snap on Ubuntu, try below:
`"arduino.path": "/snap/bin"`

It works for me.

ian185987, 2021-07-11:
Only for the ones that can't even solve this. What I have to do was to use a file from Arduino, the ones where you save your projects and add that one as "arduino.path"
[image]

gone2808, 2021-07-13:
How to add path if I dont install Arduino. I just want to install a older version which is .rar file

Grzybenc69, 2021-08-30:
I had problem with that but i find a solution.
If you are on windows instead of "arduino.path": "C:\\Program Files (x86)\\Arduino", write "arduino.path": "C:\\Program Files (x86)",

gone2808, 2021-08-30:
@Grzybenc69  yuh, I have fixed that. but 1 more problem is when I reset vscode, the path is reset to default - which make all added lib path and also arduino.path  disapear

JonasPinto, 2021-10-05:
> Para sua informação: A partir de agora, existem 2 IDE do Arduino disponíveis: 1.8 e 2.0 beta Para informá-lo, o 2.0 beta não funciona com o plugin VS Code. Você deve usar a versão 1.8.
> 
> Tentei usar 2.0, mas sem sorte. Depois de baixar o 1.8 e atualizar as configurações do VS Code (arduino.path), o plug-in parou de reclamar que não conseguia encontrar o IDE do Arduino.

estou utilizando a versão 1.8.16, mesmo assim ainda retorna essa msg

Jekahome, 2021-11-08:
Example Path: /home/jeka/Programs/arduino-1.8.16/arduino

    "arduino.path": "/home/jeka/Programs/arduino-1.8.16",
    "arduino.commandPath": "arduino"

Francisco-Castillo, 2022-01-22:
Muchas gracias, me funciono.

Francisco-Castillo, 2022-01-22:
> If you install Arduino IDE using Snap on Ubuntu, try below: `"arduino.path": "/snap/bin"`
> 
> It works for me.

Funcionó, muchas gracias!

jaindinkar, 2022-01-30:
> Example Path: /home/jeka/Programs/arduino-1.8.16/arduino
> 
> ```
> "arduino.path": "/home/jeka/Programs/arduino-1.8.16",
> "arduino.commandPath": "arduino"
> ```

This works for me on Ubuntu 20.04, Arduino snap version is a bit tedious to deal with as it creates a sandbox environment and is difficult to provide a path for, at least in my case it was not working, I rather downloaded the precompiled arduino-cli [v0.20.0](https://github.com/arduino/arduino-cli/releases/tag/0.20.2), extracted and placed it inside <kbd>/home/user-name/apps/</kbd> then provided paths as described above. Also, if you are using the cli, don't forget to check the **Use Arduino Cli** in extension settings.

kiambati-maxwel, 2022-02-02:
> 

it works

HappierGore, 2022-02-16:
For Windows:
I've installed Arduino IDE from internet, NOT FROM STORE.
After that, I used this config:

```
"arduino.path": "C:\\Program Files (x86)\\Arduino",
"arduino.commandPath": "arduino.exe"
```

And now it's working :)

dnettoRaw, 2022-03-03:
for Mac 
`
"arduino.path": "/Applications",
"arduino.commandPath": "/Contents/MacOS/arduino"
`

chinanoahli, 2022-03-11:
Is this setting(`"arduino.path"`) has to be `C:\\Program Files (x86)\\Arduino` in windows?
My arduino install path is `D:\arduino-1.8.16\arduino.exe`, and I put this in my setting.json, it never work:
```
"arduino.path": "D:\\arduino-1.8.16",
"arduino.commandPath": "arduino.exe",
```

arrazolahn, 2022-03-19:
> For Windows: I've installed Arduino IDE from internet, NOT FROM STORE. After that, I used this config:
> 
> ```
> "arduino.path": "C:\\Program Files (x86)\\Arduino",
> "arduino.commandPath": "arduino.exe"
> ```
> 
> And now it's working :)

thanks this worked!

Surendra-Srinivas, 2022-06-04:
> for Mac
> `"arduino.path": "/Applications", "arduino.commandPath": "/Contents/MacOS/arduino"`

Hey @dnettoRaw its not working. I did as you said, but the problem still persists. Can you please tell what to do...
[image]

dnettoRaw, 2022-06-04:
> > for Mac
> > `"arduino.path": "/Applications", "arduino.commandPath": "/Contents/MacOS/arduino"`
> 
> Hey @dnettoRaw its not working. I did as you said, but the problem still persists. Can you please tell what to do... [image]

hey, to many "," in your config file, maybe is that
[image]

Surendra-Srinivas, 2022-06-04:
@dnettoRaw Even if I remove the "," there, it is still throwing an error...
[image]

dnettoRaw, 2022-06-04:
@Surendra-Srinivas do you try to reinstall your Arduino IDE ? [download from Arduino.cc](https://www.arduino.cc/en/software)
and copy/paste Arduino.app to /Application (not same as /Users/{YOUR_USER_NAME}/Applications), I remember in my first try, I can't make this work because a have a 'bad' Arduino version

Surendra-Srinivas, 2022-06-04:
> @Surendra-Srinivas do you try to reinstall your Arduino IDE ? [download from Arduino.cc](https://www.arduino.cc/en/software)
> and copy/paste Arduino.app to /Application (not same as [...]

@dnettoRaw Thank you so much. It worked ! 🥇 💯

vhdlrocks, 2022-07-08:
VScode version: 1.69.0
Win10
Arduino extension for VScode: 0.4.12
C/C++ extension for VScode: 1.11.1

I had a lot of problems with this and am not a power user, so.... The Arduino CLI setting is turned on by default (didn't know the CLI was a standalone tool), see VScode command pallet: Extensions > Arduino configuration

Turn this off if your not using it and everything should work just fine. The CLI tool (arduino-cli.exe) can be found [here ](https://github.com/arduino/arduino-cli/releases/tag/0.13.0)if you want it.

CoryC3K, 2022-07-29:
After trying literally every suggestion above, vhdlrocks cued me in to the solution.

Windows 10, VSCode, Arduino IDE.

1. Install Arduino 1.8-ish from the official website (not v.2+)
2. Open Arduino IDE and ensure libraries and boards are up-to-date
3. Install Arduino VSCode extension
4. Edit settings.json

settings.json should have the following arduino settings:

    "arduino.path": "C:\\Program Files (x86)\\Arduino",
    "arduino.commandPath": "arduino.exe",
    "arduino.useArduinoCli": false,

5. Re-load the VSCode window (hint: F1 for command palate, then type Developer:Reload Window)
6. Open a *new* folder
7. Open the Command Palate again, but this time type: "Arduino: Initialize"

If that works, you're golden. Hook up your usb, set the programmer and such in the bottom right, then use the upload button up by the file tabs. 

That said, devlopment here still is terrible. It routes everything through the "Output" console, so (for example) Leaving a serial monitor up while you're building just cannot be done.

VSCode is great. Arduino is great. They don't work well together.

Juanc179, 2022-08-29:
Hi, i tried 

> After trying literally every suggestion above, vhdlrocks cued me in to the solution.
>
> Windows 10, VSCode, Arduino IDE.
>
> 1. Install Arduino 1.8-ish from the official website (not v.2+)
> 2. Open [...]

hi, i don't have arduino in that directory(windows 11). i just downloaded arduino and installed it but the directory in inside c:/program files/arduino_1.18.16.
i tried with that route but the problem "not detecting arduino.h" persist. 
i also have the latest rc build of arduino IDE. do you recommend reinstalling all arduino versions? (when i downloaded the 1.18, after alredy having the rc version, it didn't generate any new directories, just launched directly. i copied the extracted files of the .zip file to the program files directory and also didn't work. 
[image]
[image]
any guess? the project was created from the ESP-IDF extension for vs code using "arduino as a component". it opened the project as you are seeing it in the first image.
thank you for your time.

hiawui, 2022-10-12:
> for Mac
> 
> ```
> "arduino.path": "/Applications",
> "arduino.commandPath": "/Contents/MacOS/arduino"
> ```

The latest Arduino IDE app name had been changed to 'Arduino IDE' from 'Arduino'. But this extension is still checking the name 'Arduino', so it should fail to check.

My solution is to use 'arduino-cli' as below which works for me.

```
"arduino.useArduinoCli": true,
"arduino.path": "/path/to/arduino-cli",
"arduino.commandPath": "arduino-cli"
```

tomdelahaba, 2022-10-13:
> 

Great starting point, thanks a lot for it, but let me clarify few things for people that are even less aware of such a things than I am :)

1) install homebrew (https://brew.sh/) (if you have, proceed to step 2)
2) install arduino-cli (https://arduino.github.io/arduino-cli/0.21/installation/) with `brew install arduino-cli`
3) find where the arduino-cli is installed. Usually (on mac) it will be /opt/homebrew/bin/arduino-cli... Which means, if you run the command arduino-cli, it will execute this script... You can find the location with:
`which arduino-cli` (I have zsh, I am not aware if it will be the same for older bash, probably it will, I am not so skilled in this one, but you can try to use `find` instead of `which`. But which is working for me
4) lets assume you have the path, for me it was `/opt/homebrew/bin/arduino-cli`
5) proceed to VScode, go to settings (well, lets assume we will be working with the json settings
6) in my case, the input is following:
```
    "arduino.useArduinoCli": true,
    "arduino.path": "/opt/homebrew/bin/",
    "arduino.commandPath": "arduino-cli"
```
Note, even when the path to arduino-cli is `/opt/homebrew/bin/arduino-cli`, we are removing the script name from the path... But we are adding this to the commandPath

zakariae87, 2022-10-25:
Should update path variable on windows , i think it should resolve issues related to vs code pop up.

animeshmaiti, 2022-11-23:
> 

only you solved this problem for new arduino user

animeshmaiti, 2022-11-23:
> [image]
> 
> Ich habe Windows 10 un bekomme es einfach nicht hin den Arduino.path anzugeben

use arduino 1.8.X vesion 2.x is not compaitable with vs code

william-green, 2022-11-25:
> Have you installed [Arduino IDE](https://www.arduino.cc/en/Main/Software)? And please add the setting of "arduino.path", if in Windows, go to Visual Studio Code, click File > Preferences > Settings, [...]

Looks like the new versions of the IDE breaks it in Windows 10. The only way I could get it working on Windows was with downloading the zip for 1.8.19. You can find it on the legacy downloads page. The newest version (2.02) does not work. Set an environment path variable for your user in the path environment variable. Copy the path to your arduino ide extracted directory into your workspace like you've done

HydrogenStudioZ, 2022-12-01:
I FIGURED IT OUT!

Download and install this version and it works. This version installs in program files and is the only one that works with VSCode. You can still use the newer version for other things but in order to work with code you need to install and point to the directory of this one. 

https://downloads.arduino.cc/arduino-1.8.19-windows.exe

ZS-Yan, 2022-12-03:
> FYI: As of now, there are 2 of Arduino IDE available: 1.8 and 2.0 beta Letting you know, the 2.0 beta doesn't work with VS Code plugin. You have to use version 1.8.
>
> I've tried to use 2.0, but no [...]

Thanks, following your suggestion, i have solved.

avdg, 2022-12-08:
The cause of the issue might be here:

https://github.com/microsoft/vscode-arduino/blob/130aa64f050dedf9bf627060e067a6cc6f41b6e6/src/common/sys/win32.ts#L31

Not sure what `arduino_debug.exe` is though

GOD-pro, 2022-12-23:
What worked for me (fedora , snap installed arduinoIDE) was to edit the arduino Path to ("arduino.path": "/snap/bin/",    ) in the json file instead of the UI in Vs Code .
At last don't forget to restart Code.

berparis, 2022-12-31:
MacOS Ventura 13.0.1 with vscode  1.74.2 
I've tried lot of things in the setting to get Arduino extension working but nothing works ... : 

**Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change.** 

here is the most logical settings I have been tried but it fails :
`{
    "arduino.path": "/Applications/Arduino IDE.app",
    "arduino.commandPath": "/Contents/MacOS/Arduino IDE",
    "arduino.enableUSBDetection": true,
    "arduino.logLevel": "verbose"
}`

note: this is the executable I can see in my terminal console with the  unix "ps" command :
"/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"

Any idea ?

berparis, 2023-01-03:
It's seems "Arduino IDE" versions 2.x are no more supported by vscode.   I've installed "arduino-cli" instead and its'ok, vscode arduino extension is runnning ok.

Splarkszter, 2023-02-07:
Linux Mint Arduino 1.8.19

Can't make it work.

Ugh i only want this because of intellisense.

NT26, 2023-02-13:
Please you try to Arduino old version, 
I'm use Window OS Arduino 1.8.19 It's work, After change from 2.0.3

quanguyen, 2023-12-08:
`function ... may not be initialized`

VS Code 1.85.0 on MacOS Sonoma 14.1.2; Arduino extension v0.6.0

After lots of configuring in C/C++ Intellisense, I got VS Code passed #include files.

Now, I got the error on every constant and variable declaration and initialization: `function ... may not be initialized` and `expression must be a modifiable lvalue`. (attached screenshot)

How should I fix this?

[image]

kameo4242, 2024-01-15:
I finally got this Arduino 2.1 + Arduino Cli + VScode working on Windows (with intellisense).
I documented everything [here](https://github.com/microsoft/vscode-arduino/issues/1541#issuecomment-1892424134).