microsoft/vscode-arduino #952: Save the last used end of line

MagTun, 2020-01-02:
Please add a way to save the last use end of line so we don't have to select it again each time. One way of doing could be via settings.json

Thanks!

hellyzh, 2020-01-06:
Thank you for your advice. Do you have more details?

MagTun, 2020-01-08:
No, I don't know how to implement this, I just know many extensions use the settings.json to store such users preferences.

hellyzh, 2020-01-08:
What do you mean the last use end of line? is it the selection after click `Arduino: Open Serial Monitor` like "No line ending", "Newline" ...

MagTun, 2020-01-08:
Yes this is what I mean:  when I want to send text to my connected device, I always have to make sure the end of line is the right one because each time I restart vscode the line ending is set back "No line ending" and not the one I usually use (ex: Newline). If my choice was saved in settings.json, then each time vscode would restart, it will automatically set the end of line to the one I want (ex: Newline and not "No line ending")

I also have to make sure the end of line is the right after I have to "restart extension host" because I unplugged a device connected to serial monitor without closing the serial monitor first (by the way. that would be great if there was a way to prevent the crashing of "extension host" in this case but this is another issue).

Xepol, 2020-02-26:
I issued a pull request (baud_fix) that covers this.  Also you to set this before opening the serial monitor.  Hope they accept it in to the main branch soon.

hellyzh, 2020-03-26:
Arduino extension 0.3.0 has been released. It removed the selection of ending line and use new line with `/r/n` as default.

closed 2020-03-27 by hellyzh

MarcoRBosco, 2021-07-09:
How do we change now the ending line? If  for example we need no line ending instead of `/r/n`. Thank you!

roninbar, 2021-08-20:
I don't understand how to select a line ending other than `/r/n`. Why is there an option in the status bar to change the baud rate but no option to change the line ending? (v0.4.3)