microsoft/vscode-arduino #954: Problem calling Arduino commands from Custom Task.

drewsed, 2020-01-08:
Hi,

first of all thanks for this great Arduino extension. 

I added a custom task "ESP32 OTA Upload" to vs code, which simply calls the `espOTA.py` script to upload the build over the air.
In order to chain the `Arduino: Verify` command with my custom Task, I used the following task definition:

```json
{
            "label": "ESP32: Build",
            "command": "${command:arduino.verify}",
            "problemMatcher": []
 }
```

after that I used the 
`"dependsOn": "ESP32 OTA Upload"`
flag, to run the task "ESP32: Build" prior to my custom task.

This is working fine!

But then I wanted to bring the SerialMonitor panel in front after the upload procedure.
To do this, I simply added two tasks:
```json
{
            "label": "ESP32: Close Serial Monitor",
            "command": "${command:arduino.closeSerialMonitor}",
            "problemMatcher": []
},
{
            "label": "ESP32: Open Serial Monitor",
            "command": "${command:arduino.openSerialMonitor}",
            "problemMatcher": []
}
```
But `${command:arduino.closeSerialMonitor}` is not working. I can open the SerialMonitor, but cant close it with this Task. If I spell the command wrong, like ${command:arduino.**C**loseSerialMonitor} (capital C ), I get an error message popping up in the right bottom corner.

I do not know how to debug this, the only thing i did was checking correct spelling of the command names in the [package.json](https://github.com/microsoft/vscode-arduino/blob/master/package.json#L46). 

I wonder why all commands are working exept `arduino.closeSerialMonitor`.

Thanks in advance!

hellyzh, 2020-01-09:
Please change the task of close Serial Monitor as below and try, it should work.
````json 
 "tasks": [
    {
        "label": "ESP32: Close Serial Monitor",
        "command": "echo ${input:closeSerialMonitor}",
        "problemMatcher": []
    }
],
  "inputs": [
    {
        "id": "closeSerialMonitor",
        "type": "command",
        "command": "arduino.closeSerialMonitor"
    }
  ]
````

drewsed, 2020-01-09:
Hi @hellyzh,
thank you very much! It is working fine now. For the future I will share my settings here:

**tasks.json:**
```cpp
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            // call verify
            "label": "ESP32: Build",
            "command": "${command:arduino.verify}",
            "problemMatcher": []
        },
        {
            // just upload the last build vial OTA
            "label": "ESP32: OTA Upload (192.168.178.47)",
            "type": "shell",
            "command": "py %LOCALAPPDATA%\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\tools\\espota.py",
            "args": [
                "-i",
                "192.168.178.47",
                "-a",
                "passwordOTA",
                "-p",
                "3232",
                "-f",
                "${workspaceFolder}\\build\\main.ino.bin",
                "-d"
            ],
            "options": {
                "shell": {
                    "executable": "cmd.exe",
                    "args": ["/c"]
                }
            },
            "problemMatcher": []
        },
        {
            // call closeSerialMonitor
            "label": "ESP32: Close Serial Monitor",
            "command": "echo ${input:closeSerialMonitor}",
            "problemMatcher": []
        },
        {
            // call openSerialMonitor
            "label": "ESP32: Open Serial Monitor",
            "command": "${command:arduino.openSerialMonitor}",
            "problemMatcher": []
        },
        // chaining tasks
        {
            // call verify and then close serial monitor
            "label": "ESP32: Build + Close Serial Monitor",
            "command": "echo ${input:closeSerialMonitor}",
            "dependsOn": "ESP32: Build",
            "problemMatcher": []
        },
        {
            // call verify and then close serial monitor and upload build via OTA
            "label": "ESP32: Build + close Serial + OTA Upload (192.168.178.47)",
            "dependsOn": "ESP32: Build + Close Serial Monitor",
            "type": "shell",
            "command": "py %LOCALAPPDATA%\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\tools\\espota.py",
            "args": [
                "-i",
                "192.168.178.47",
                "-a",
                "passwordOTA",
                "-p",
                "3232",
                "-f",
                "${workspaceFolder}\\build\\main.ino.bin",
                "-d"
            ],
            "options": {
                "shell": {
                    "executable": "cmd.exe",
                    "args": ["/c"]
                }
            },
            "problemMatcher": []
        },
        {
            // call verify and then close serial monitor and upload build via OTA and open serial monitor
            "label": "ESP32: Build + close Serial + OTA Upload (192.168.178.47) + open Serial",
            "dependsOn": "ESP32: Build + close Serial + OTA Upload (192.168.178.47)",
            "command": "${command:arduino.openSerialMonitor}",
            "problemMatcher": []
        }
    ],
    "inputs": [
        {
            "id": "closeSerialMonitor",
            "type": "command",
            "command": "arduino.closeSerialMonitor"
        }
      ]
}
```
It would be better to set some values via variables, but as I do not need to change values (e.g. the IP) very often, it's ok for me.

To run the task from keyboard shortcuts, i edited my **keybindings.json** file:

```cpp
// Place your key bindings in this file to override the defaults
[
    {
        "key": "ctrl+alt+l", // Schortcut Ctrl+Alt+L
        "command": "workbench.action.tasks.runTask",
        "args": "ESP32: Build + close Serial + OTA Upload (192.168.178.47) + open Serial"
    }
]
```

closed 2020-01-09 by drewsed