microsoft/vscode-arduino #956: Compile error when included header file name starts with letter earlier in the alphabet than main sketch (.ino) file (macOS Mojave)

abdirk, 2020-01-14:
### Description
when main sketch file includes header file with name starting with letter which is earlier in the alphabet than the letter the main sketch filename starts with, then the compilation fails. All header files with names starting with later letters do not cause this issue. 
### Steps to reproduce

1. open a new folder in VS Code
2. create file **main.ino** with following contents 
`#include "Arduino.h"`
`#include "Esp.h"`
`#include "aaa.h"`
`#include "rrr.h"`
`void setup() {}`
`void loop() {}`

3. create **aaa.h** and **rrr.h** files. These files may be empty
4. initialize Arduino plugin (Arduino: Initialize) for ESP 8266 board (NodeMCU 1.0 /ESP 12E)
5. Compile (Arduino: Verify)
Compilation ends with following error:
`main:4:17: error: aaa.h: No such file or directory`

when line `#include "aaa.h"` is removed compilation ends with success ("rrr.h" is found and included)

### Additional info

- Operating System: macOS Mojave, version 10.14.6 (18G1012)
- VS Code version: 1.41.1
- vscode-arduino version: 0.2.29
- Arduino IDE version: 1.8.9
- esp8266 by ESP Community (package for Arduino IDE) version: 2.6.2

Xring-git, 2020-03-22:
Here the same, but I've note that this error appears every time an Header file is the first file in alphabetical order.

I use MyDefines.h (without .ino or .cpp file)

unprime2, 2020-04-11:
I also have the same issue. When look at build folder, I see [name].hpp file became [name].hpp.cpp. That's why it said "No such file or directory". Thanks for your investigation, I decided to add "_" prefix to ino file while waiting a solution :)))

adiazulay, 2021-08-03:
Does this happen if you try to compile it using the Arduino IDE or CLI?

HowardHonig, 2021-08-03:
Adiazulay, 
Thanks for asking.  Haven't tried CLI but Arduino IDE works like it should.

adiazulay, 2021-08-03:
Is the folder containing the sketch named the same as the sketch?

I can only repro the issue if the sketch and folder have different names.

HowardHonig, 2021-08-04:
Adiazulay, thanks for helping.

I've repro it a number of times in a number of ways.  I've rebuilt intellisense a number of times.  In this example I've removed sammyDevice.h to a subfolder only to have sd.h the target of the "No such file or directory" error on the header.  I've tried adding the working folder in the search path.  I've rebooted and started VSC a number of times.  I've uninstalled VSC and reinstalled it even deleting the vsc setups.  Nothing helps except using subfolders.     

Here is a sample that can't get much simpler.  Please note that using the #ifndef to make sure the header doesn't get include 2x was just added to make sure the purest does point that out.  It didn't matter if it's added or not.  The #defines in the header are just placements so the file isn't empty.  Doesn't matter if it's real code.

file **sammyDevice.ino**  ------>
#include <arduino.h>
#include "sd.h"
#include "sammyDevice.h"

void setup() {
    Serial.begin(115000);
    Serial.printf(WHICHOME);
}

void loop() {
}

file **sammyDevice.h**  ----->

#ifndef ANYTHING_H
#define ANYTHING_H 1

#define ANYTHING anything

#endif

file  **sd.h**  ->>>>>
#ifndef _SAMMYDEVICE_h
#define _SAMMYDEVICE_h

#define     WHICHOME        "whichone"

#endif