microsoft/vscode-arduino #957: Unable to Debug Adafruit SAMD Boards

PTS93, 2020-01-20:
Whenever I try to start a debugging session with any Adafruit based SAMD21 board I get a pop up saying that its currently not supported, even the officially supported board the Feather M0 gives the same error message.
Is it not possible to debug any generic SAMD21 board without it being tested? They all behave the same way.

GregTerrell, 2020-05-06:
I don't know about your full development environment, but I have had success with the SAMD21 (Adafruit Feather M0 Express) and wrote an Medium article with the setup and steps. Have YouTube and supporting files on github as well. 

Hope this helps,
Greg

robotdad, 2020-10-30:
@GregTerrell could you share a link to that article? I wasn't able to find it.

GregTerrell, 2020-10-30:
You can find it on the LooUQ Blog [here](https://www.loouq.com/post/developing-embedded-samd-applications-with-vs-code-arduino). It is also on [Medium](https://medium.com/@Greg_LooUQ/developing-embedded-applications-with-vs-code-arduino-68eaa232643)