microsoft/vscode-arduino #959: Update board type command does not update intellisense config automatically

whiteb3ar, 2020-01-25:
When board type changed "defines" section in c_cpp_properties.json is not updated.

When compiling the code "-mmcu={board}" switch is passed as arg so that there is correct "#define" whilst compilation and there is no errors. But intellisense has no correct "#define" available until manually specified.

Actual result
[image]

Expected result
Intellisense highlight correct code block

whiteb3ar, 2020-01-26:
[This one](https://github.com/Microsoft/vscode-cpptools/issues/2377) causes false positive intellisense errors as well.

elektronikworkshop, 2020-02-04:
I have the same issues. The only proper way to do it is to parse the build output and auto fill the intellisense settings. But when looking at the push activity here I get the impression that this project is more or less dead.

elektronikworkshop, 2020-02-05:
Hi to everyone who is suffering from this!

I was so annoyed, that I began to work on an auto-configuration mechanism, which parses the build output from Arduino and creates a `c_cpp_properties.json` from it. The goal is that no one wants likely ever edit this file manually again - except for very weird setups for which I will add a configuration flag/option that will keep the extension from touching the file.

I coded some proof of concept already but I haven't committed anything yet. But I began documenting the project [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md).

Anyone who wants to keep me motivated can chip in some :beer:-money - the donation button can be found in the link above.

Other support is appreciated and comprises
* people with Windows and OSX setups to test the cross platform compatibility
* the original maintainers could help me with CI and how to build test-releases

The progress can be followed within the same file. The [repo](https://github.com/elektronikworkshop/vscode-arduino) is set to this feature branch by default.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay