microsoft/vscode-arduino #962: Compilation unable to find include file

jnanneng, 2020-02-07:
Intellisense and the IDE resolve the include files just fine, however when compiling an INO file, an error is encountered that an include file cannot be found.  Screenshot attached.
[image]

elektronikworkshop, 2020-02-08:
Hi,

this doesn't seem to be a problem of vscode-arduino. You try to develop a library for Arduino but it looks like you don't know how to do it. Read the [documentation on how to write Arduino libraries](https://www.arduino.cc/en/Hacking/libraryTutorial) first. There's even more to it than this document mentions -> use google.

As a general advice: Please try to compile your projects in the Arduino IDE first. If it doesn't compile there, you must not file an issue here.

Regards, EW

Maintainers: This issue can be closed.

hellyzh, 2020-02-21:
Thanks @elektronikworkshop for explaining. And the file c_cpp_properties.cpp is only used for intellisense config not compiling include path.

elektronikworkshop, 2020-02-21:
You're welcome!

closed 2020-03-26 by hellyzh