microsoft/vscode-arduino #966: Express server is started on random port when vscode-arduino is activated

mathieubrun, 2020-02-11:
## VSCode info
Version: 1.41.1
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-20T11:11:01.597Z
Electron: 6.1.7
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 5.5.2-arch2-2

## Extension version
0.2.29

## Description:
A random port is opened on the machine on which the extension is running, with an Express server listening. The opened port is available form other hosts on the same network.

## Steps to reproduce (linux host) :
1. install extension
2. run `sudo netstat -tulpan | grep LISTEN` and note the port (here 44873)
````
tcp6       0      0 :::44873                :::*                    LISTEN      16511/electron      
````
3. run `curl localhost:44873` (replacing 44873 with port noted in 2.
````
<html>
  <head>
    <style title="themestyle"> </style>
    <script type="text/javascript">
      // inherit theme styles from vscode ide
      var decode = function (s) {
        return decodeURIComponent(s.replace(/\+/g, " ")); 
      };
      var getStyleSheet = function (unique_title) {
        for(var i=0; i<document.styleSheets.length; i++) {
          var sheet = document.styleSheets[i];
          if(sheet.title == unique_title) {
            return sheet;
          }
        }
      };
      var match,
          search = /([^&=]+)=?([^&]*)/g,
          query  = window.location.search.substring(1),
          queryParams = {};
      while (match = search.exec(query)) {
        queryParams[decode(match[1])] = decode(match[2]);
      }
      var supportedThemes = ["vscode-dark", "vscode-light"];
      var theme = "vscode-dark"; // default theme.
      if (supportedThemes.indexOf(queryParams["theme"]) > -1) {
        theme = queryParams["theme"];
      }
      var bgcolor = queryParams["backgroundcolor"];
      var color = queryParams["color"];
      var stylesheet = getStyleSheet("themestyle");
      stylesheet.insertRule(".vscode-theme { background-color: " + bgcolor + "; color: " + color + "; }", 0);
      stylesheet.insertRule(".vscode-theme .theme-bgcolor { background-color: " + bgcolor + "; }", 1);
      stylesheet.insertRule(".vscode-theme .theme-color { color: " + color + "; }", 2);
      stylesheet.insertRule("html > body { background-color: " + bgcolor + "; color: " + color + "; }", 3);
      document.documentElement.className = theme + " vscode-theme";
    </script>
  <link href="styles/app.css" rel="stylesheet"></head>
  <body>
    <div id="mainContent"></div>
  <script type="text/javascript" src="app.js"></script></body>
</html>
````
4. disable extension
5. run `sudo netstat -tulpan | grep LISTEN` again
> nothing

## Expected behavior

No port is opened, or at least opened on a local only mode.

Culprit seems to be this file : https://github.com/microsoft/vscode-arduino/blob/master/src/arduino/localWebServer.ts

hellyzh, 2020-02-19:
This is by design. It should not cause security issue.

mathieubrun, 2020-02-28:
What is the purpose of this feature ? Could you point me to its documentation ?

elektronikworkshop, 2020-02-29:
Thanks Mathieu for this detailed report.

This is definitely a security issue IMHO. Should rather be something like
```ts
listen(0, "localhost")
```
(without diving too deep into the reference code)

I take a look at it tomorrow...

Regards,  
EW

elektronikworkshop, 2020-02-29:
Hi Mathieu,

looks better now:
```
tcp        0      0 127.0.0.1:13248         0.0.0.0:*               LISTEN      5452/code           
tcp        0      0 127.0.0.1:60048         0.0.0.0:*               LISTEN      5484/code           
tcp        0      0 127.0.0.1:46865         0.0.0.0:*               LISTEN      5390/code           
tcp        0      0 127.0.0.1:47859         0.0.0.0:*               LISTEN      5390/code
```
I had to rewrite the whole web-server initialization, as they didn't even start it asynchronously and didn't wait until the socket was ready and listening. This is required for host resolution though...

You can test-drive it - it is part of my alpha18 release and available from my [alpha release Dropbox folder](https://www.dropbox.com/sh/whmcdt26chyjgby/AAB1Ld2fzZ9Z_NfM3CRay17wa). 

I will file a pull request will all my other work - the most important being the proper IntelliSense support.

Thanks again,  
Regards,  
EW
____
Microsoft -- *security issue by design*<sup>TM</sup>  
:wink: