microsoft/vscode-arduino #967: Add command to burn bootloader

vlastahajek, 2020-02-12:
This is a continuation of closed #864 .

Arduino IDE contains a command `Burn Bootloader`. This is probably not often used for classic Arduino boards with bootloader inside, but it is an important command when using single chips without a bootloader (e.g. ATtiny) to write fuses.

elektronikworkshop, 2020-02-15:
As vscode-arduino requires the Arduino IDE anyways. Why don't you burn it from there?

vlastahajek, 2020-02-17:
Sure, this the only way how to do it now. 
However,  why to open another IDE, when all other functions are available in VS Code through this plugin?
This issue is an enhancement, not a bug.

elektronikworkshop, 2020-02-17:
Yes, of course it's a feature request. But IMHO vscode-arduino must not implement all functionality the original IDE implements. Burning a bootloader is very architecture specific (it's not even supported by the arduino CLI which is the primary interface vscode-arduino makes use of) and it is a task that a regular user doesn't encounter frequently during his daily use of vscode-arduino. The implementation in contrast is pretty complex with a lot of corner cases. Therefore I don't think this will be implemented unless someone implements it himself and files a pull request (as I currently do with proper IntelliSense support and a better serial monitor later).
Regards!

THX2112, 2020-06-11:
Seconding this feature. VSCode currently has all it needs to implement this as far as I can tell.

Anyone doing development on bare controllers has to flash the bootloader many times, sometimes every time a new compile is done as I'm doing now.

Stady234, 2021-01-14:
Is there any updates on this? is it still being considered?  Would be great to not have to have ArduinoIDE just for burning a bootloader... then use VSCODE for everything else...

davidcooper1, 2022-02-19:
Based on the `arduino` man page, this feature is not exposed through command line arguments. However, `arduino-cli`'s `burn-bootloader` command seems easy enough to use. An upstream request could be made to the [Arduino IDE Repo](https://github.com/arduino/Arduino), but until then, this would probably have to be an `arduino-cli` only feature.

[arduino Reference](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc)
[arduino-cli Reference](https://arduino.github.io/arduino-cli/0.21/commands/arduino-cli_burn-bootloader/)