microsoft/vscode-arduino #969: INO Files defines undefined but can jump to definition

rnd-ash, 2020-02-16:
Within Arduino INO files, values that are defined in project level header files make intellisense throw the following error:
[image]

However, as shown, it appears it is able to jump to definition of the define. Project compilation also works fine without an issue

elektronikworkshop, 2020-02-16:
vscode-arduino does not configure IntelliSense correctly. This is a duplicate of #438 (and many others). I'm currently working on this issue (I'm not associated with the original maintainers though). A beta test should become available in the forthcoming week. See the issue above for more information.

Before filing an issue, please search the existing issues first. This way we can focus on the stuff that matters. Thanks!

Maintainers: Can be closed as duplicate.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay