microsoft/vscode-arduino #970: Extension issue

acinsa, 2020-02-18:
- Issue Type: `Bug`
- Extension Name: `vscode-arduino`
- Extension Version: `0.2.29`
- OS Version: `Windows_NT x64 10.0.16299`
- VSCode version: `1.41.1`

:warning: We have written the needed data into your clipboard. Please paste! :warning:
```json 
{
	"messages": [],
	"activationTimes": {
		"codeLoadingTime": 485,
		"activateCallTime": 434,
		"activateResolvedTime": 0,
		"activationReason": {
			"startup": true,
			"extensionId": {
				"value": "vsciot-vscode.vscode-arduino",
				"_lower": "vsciot-vscode.vscode-arduino"
			},
			"activationEvent": "*"
		}
	},
	"runtimeErrors": [
		{
			"name": "TypeError",
			"message": "Cannot read property 'map' of undefined"
		}
	]
}
```

hellyzh, 2020-02-19:
Do you have more details and steps about how to reproduce this issue?

hellyzh, 2020-03-26:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-03-26 by hellyzh