microsoft/vscode-arduino #974: customBuildProperties does not put quotes on build.path

SlavDimov, 2020-02-29:
Ext version: 0.2.29
VScode version: 1.42.1
Arduino IDE version: 1.85
Board selected: one of the [STM32Duino boards ](https://github.com/stm32duino/Arduino_Core_STM32)

The contents of my arduino.json look like:
```
{
    "sketch": "...",
    "board": "...",
    "configuration": ...,
    "output": "build"
}
```

However the full path of the output folder is
```
/home/myname/a_dir/another dir with spaces/myproject/build
```

The sketch verification fails with the error:
```
....
bash: line 0: [: too many arguments
mkdir: cannot create directory ‘another’: Permission denied
mkdir: cannot create directory ‘dir’: Permission denied
mkdir: cannot create directory ‘with’: Permission denied
mkdir: cannot create directory ‘spaces’: Permission denied
....
```

When I look at the build/build.options.json:
```
...
"customBuildProperties": "build.path=/home/myname/a_dir/another dir with spaces/myproject/build,...
...
```
Notice how the path is not in quotes. I suspect this is what is causing the issue. However I do no know how to force the arduino.json to put quotes around it.

elektronikworkshop, 2020-02-29:
Hi Slav,

thanks for reporting!

This seems to be a bug. I'll fix this in my fork and provide you with a link to an alpha version later.

Regards,  
EW

SlavDimov, 2020-02-29:
Hi EW,
thanks for the quick reply!

Waiting on the fix!

Regards,
Slav

elektronikworkshop, 2020-02-29:
You're welcome!

I can't reproduce the problem with an Arduino Nano. Here's my build output:
```
[Starting] Verify sketch - app.ino
Please see the build logs in Output path: /home/uli/Projects/delme with spaces/yet another folder/build
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
<...snip...>
```
And my configuration
```json
{
    "board": "arduino:avr:nano",
    "configuration": "cpu=atmega328",
    "sketch": "app.ino",
    "output": "build"
}
```
But with a BluePill I'm able to reproduce. This leads me to the conclusion, that STM32Duino's build back-end can not handle output paths with spaces. You should file an issue there (and close this one :wink: of course).

Regards,
EW

P.S.: The reason I thought it was a bug: I'm currently implementing [better IntelliSense support](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md) for this extension and thought I spotted something (I'm pretty familiar with the code base by now). But I was wrong.

SlavDimov, 2020-02-29:
Hello EW, 
Interesting result, I did not bother to try with other board types once I found the missing quotes. 
Thank you for your time. I will open an issue in the stm32duino repo.

closed 2020-02-29 by SlavDimov